dotnet/iot #1123: Hardware PWM issue

fabus1989, 2020-07-02:
Hi,

I have problem with hardware PWM, cannot run it from .net core application - LED connected to GPIO12.
PWM is enabled in /boot/config.txt file (dtoverlay=pwm-2chan)
Sample python script works ok.

What I'm missing?

Here is the code sample:
`﻿using System;
using System.Device.Pwm;

namespace Sample
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");
            var pwm = PwmChannel.Create(0, 0, 100, 1);
            pwm.Start();
            Console.ReadKey();
        }
    }
}`

 I'm using Raspberry PI 4 with 2GB of ram.

Best regards
Fabian

joperezr, 2020-07-02:
Hello thanks for logging the issue. I believe that the Raspberry Pi doesn't support frequency as low as 100, can you instead try something like `10000` for frequency and perhaps a `.5` for dutyCycle? dutyCycle will work fine with a value of 1 but the idea would be to dim it so that you can notice the change of brightness.

pgrawehr, 2020-07-02:
I've also observed issues when the frequency is below 10000 (or so). I'm not sure what's going on then, but it starts blinking with the given duty cycle, but at frequencies of like 1 Hz. I don't know whether this is a hardware or a software issue, though.

krwq, 2020-07-03:
is it possible you got multiple app instances running on the machine? I've had this happen to me couple of times when working over ssh

fabus1989, 2020-07-05:
@joperezr I've tried with multiple frequecy values (100,400,1000,10000) and with few duty cycle (0.25, 0.5, 0.75, 1.0). Unfortunatelly LED light is always off. I do not have multiple instances running. I also tried with different channels and different pins. In config.txt pwm setting is at the bottom of a file. Do you have other ideas what to check?

fabus1989, 2020-07-05:
One strange thing I observed, usually first time (after raspberry reboot) when I run program I'm receiving exception.
``Unhandled exception. System.UnauthorizedAccessException: Access to the path '/sys/class/pwm/pwmchip0/pwm0/duty_cycle' is denied.
 ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)
   at System.IO.FileStream.OpenHandle(FileMode mode, FileShare share, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access)
   at System.Device.Pwm.Channels.UnixPwmChannel..ctor(Int32 chip, Int32 channel, Int32 frequency, Double dutyCycle)
   at System.Device.Pwm.PwmChannel.Create(Int32 chip, Int32 channel, Int32 frequency, Double dutyCyclePercentage)
   at Test.Program.Main(String[] args) in C:\Users\fabia\source\repos\Test\Test\Program.cs:line 11``

krwq, 2020-07-05:
@fabus1989 what version of System.Device.Gpio are you using? Can you share your csproj and Nuget.config if you have one?

fabus1989, 2020-07-05:
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.1</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="Iot.Device.Bindings" Version="1.0.0" />
    <PackageReference Include="System.Device.Gpio" Version="1.0.0" />
  </ItemGroup>

</Project>
```

pgrawehr, 2020-07-06:
Please try again against the latest dev nuget. This should solve the access violation problem at least.

Ellerbach, 2020-08-24:
@fabus1989, did updating the nuget packages helped?
Please refer to the 2 command lines explained in the main readme of the repo to do it: https://github.com/dotnet/iot#net-cli
You can as well use Visual Studio, add the URL https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json to the list of nuget source and select it from Visual Studio when you search for a nuget

SavaNDragos, 2020-08-25:
Run over this issue, so I updated to:
    <PackageReference Include="System.Device.Gpio" Version="1.1.0-prerelease.20276.1" />
That "System.IO.IOException: Permission denied" bug is fixed.

Ellerbach, 2020-08-26:
> That "System.IO.IOException: Permission denied" bug is fixed.

Thanks, good news. I guess we will then be able to close this issue.

fabus1989, 2020-08-26:
So basically yes, upgrading nuget to newer version solved problem with "System.IO.IOException: Permission denied", but not at all.

PWM, still not work as expected. I found workaround and now, before I run app, I setting PWM mode one the pin using external WiringPI library.

So it looks, that when you creating PwmChannel, using .net core iot library, it isn't checking if pin is in PWM mode.

I've tried to create some solution for that, but do not have enought time and skill to do it right now.

pgrawehr, 2020-08-26:
@fabus1989 : That's actually expected. This library currently doesn't change any pin modes (except input/output). You can add the following line to `/boot/config.txt` to enable pwm at startup:
```
dtoverlay=pwm-2chan,pin=12,pin2=13,func=4,func2=4
```
the pins can be 12 and 13 or 18 and 19. With wiringpi you can check the state of these pins after startup (run `gpio readall`. The pins (12 and 13 for me) should say "ALT0" in the "mode" column. 

FYI: There's a concept for updating pin modes in discussion. See #1128

Ellerbach, 2020-08-26:
@fabus1989, as @pgrawehr mention, it's what is expected, so you gave me the idea to add more documentation on how to properly activate it. Check PR #1168. That should help you! Feedback more than welcome

krwq, 2020-11-24:
@fabus1989 I'm closing the issue given we test hardware PWM on CI and @Ellerbach has updated docs to make it easier. Please feel free to re-open if you feel there is anything else we can improve.

closed 2020-11-24 by krwq