dotnet/iot #1127: .NET Core LTS Branch

djaus2, 2020-07-03:
**Is your feature request related to a problem? : No.**

Can I suggest that a clear path to the version of the repository for the current LTS version of .NET Core.
Current LTS is Version 3.1
Branch of the repository: release/3.0
The .NET Core version of the Master branch as per global.json is 5.0.100-preview.
Would be useful to show how to get it:
```git clone https://github.com/dotnet/iot.git  --branch  release/3.0```

Perhaps a paragraph in the README or a link to a Wiki page from there.

=================
Comment: I had used the repository a few months ago but found that when I came back to it in today I ran into all sorts of problems building my apps as the repository required the .NET 5 version. The LTS (long term support) version needs better support.

pgrawehr, 2020-07-03:
I think the repository only needs .NET 5 for building. The target framework for the release package is still netcoreapp2.1. 

Information about building is here: https://github.com/dotnet/iot/blob/master/Documentation/CONTRIBUTING.md (Ok, that's not 100% up to date as well...)

I do agree that we'll need to clarify versioning. The current release package is tagged 1.0.0, but the old backlog here talks about 3.0.

djaus2, 2020-07-03:
But .NET 5 is not anywhere LTS. The other issue I found is that the sample apps have changed substantially.  On my system I am using .NET Core 3.1 , with updates which I would suspect may / most would still be using. So if they try to use the Main branch of the repository then things will fall over. Here, its not like normal use of preview features with software as hardware is involved and things are therefore more trying. We need a clear unambiguous path for LTS .NET Core users to use the repository. Yes the sample do come out as .NET Core 2.1 but it is a simple matter to change that to .NET Core 3.1 in project files and . The same cannot be said for the Main branch samples.

djaus2, 2020-07-04:
OK, I've been looking at this some more today. Its not a simple matter to change from .NET Core 3.1 to .NET Core 2.1 with branch release/3.0.  I have been trying to build the repository in RPi using the build.sh command in the root. If I run it as is it fails on the nuget package builds. Also global.json is set to V3.1 whereas all of the samples etc are set to V2.1. ???
Ok so if you use the Main branch with .NET5 all should be well (tomorrow's 2Do)?
But we do need a clear path for samples with LTS versions of .NET Core.

pgrawehr, 2020-07-04:
I haven't trying to build the package on the Pi for a long time, since I never got that to work. Individual projects work, but not the whole package, due to some problems in the build system it seems. 

Why do you need the samples to be built against an LTS version? They are just here for demonstration and show the usage of some bindings. If you use the library on your own project, you can use any runtime version you want.

djaus2, 2020-07-05:
**_In Windows 10 Context:_**

Update on this: I did a complete build with the **Master Branch** of the repository unadulterated. That worked OK.
The sample apps (nearly) all are .NET Core V2.1. I  modified things as follows:
- Changed sample apps to V3.1
- There were Some V2.0, 3.0 which I changed to V3.1 as well
- Left **global.json** as is ,except changed dotnet/x64 to V3.1.5:
```
    "runtimes": {
      "dotnet/x64": [
        "3.1.5"
      ]
```
Reran build but had problems with the project **Mpu9250**:
```
C:\Users\DavidJones\source\dotnet\iot\.dotnet\sdk\5.0.100-preview.6.20310.4\Microsoft.Common.CurrentVersion.targets(2084,5): error MSB3245: Could not resolve this reference. Could not locate the assembly "System.Numerics.Vectors". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors. [C:\Users\DavidJones\source\dotnet\iot\src\devices\Mpu9250\Mpu9250.csproj]
C:\Users\DavidJones\source\dotnet\iot\.dotnet\sdk\5.0.100-preview.6.20310.4\Microsoft.Common.CurrentVersion.targets(2084,5): error MSB3243: No way to resolve conflict between "System.Numerics.Vectors, Version=4.1.6.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" and "System.Numerics.Vectors". Choosing "System.Numerics.Vectors, Version=4.1.6.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" arbitrarily. [C:\Users\DavidJones\source\dotnet\iot\src\devices\Mpu9250\Mpu9250.csproj]
```
I removed that project from the tree and the build ran OK this time.
- I was able to separately create that project's Sample project  using the 2 Nugent packages (Had to sue the latest versions, the required device doesn't exist in V1.0.0.
==============
So I do have a version of the repository that builds the Samples as .NET Core 3.1, and runs the tests OK, except for that one project

djaus2, 2020-07-06:
I have deposited my version at:  
[djaus2/iot](https://github.com/djaus2/iot/blob/master/README.md}  
Includes a list of changes.

djaus2, 2020-07-09:
After discussion with Rich and Jose its been decided to maintain compatibility with .NET Core 2.1 with the System.Devices.GPIO and Iot.Device.Bindings and with .NET Standard 2.0. 
But to upgrade all samples and Tests to .NET Core 3.1 

[djaus2/iot ](https://github.com/djaus2/iot) has now been modified to implement those features.

Ellerbach, 2020-09-15:
@djaus2 , I feel like this issue is resolved. Feel free to reopen it if you feel it's needed!

closed 2020-09-15 by Ellerbach