dotnet/iot #1132: Adafruit.BME280 always reports same values (.NetCore 3.1 MVC)

D4rkF4ce, 2020-07-08:
Heho Guys!
Since switching to iot.device.bindings, I can no longer get values from my Adafruit.BME280 sensor.

I use ASP .NetCore 3.1 MVC on a Raspberry Pi 4 with RaspberryPi OS 4.19.118

It basically worked with my UWP app running on winiot raspberry Pi 3. Now I want to switch to an ASP .NetCore MVC application, but I cannot get correct values. 

My BME280 reports everytime the same values (eg. Temp is always 21,659415762219577)

My Code in MVC Controller:

```
public string GetTemp()
        {
            string returnVal = "";

            try
            {
                var settings = new I2cConnectionSettings(1, Bme280.DefaultI2cAddress);                
                using (var i2c = I2cDevice.Create(settings))
                {
                    var sensor = new Bme280(i2c);
                    sensor.SetPowerMode(Bmx280PowerMode.Forced);                   
                    Temperature tmp = sensor.ReadTemperatureAsync().Result;
                    returnVal += "Temp: " + tmp.Celsius.ToString() + Environment.NewLine;
            }
            catch (Exception ex)
            {
                returnVal = ex.Message + Environment.NewLine + ex.StackTrace;
                if (ex.InnerException != null)
                {
                    returnVal += Environment.NewLine + ex.InnerException.Message + Environment.NewLine + ex.InnerException.StackTrace;
                }
            }

            return returnVal;
        }
```

I also try to call it async, in a loop and yeah nothing changes... :(

Any Ideas?

Many Thx!
D4rk!

pgrawehr, 2020-07-09:
Weird. I only have a BMP280 here, but that works just fine. 

Which version of the library are you using? There might be significant improvements from the official release (1.0.0) and the current development version.

D4rkF4ce, 2020-07-10:
I can't understand it too... I use version 1.0.0 of iot.bindings and ASP.Net Core 3.1 MVC and VS 2019. 
On Pi3 with the adafruit uwp library i get values... but after switching to Pi4 and iot.bindings it stops working.

Without this sensor, my entire project is now useless :( as long as i have no solution, as workaround i think i will switch to the shelly iot H&T sensor via wifi or cloud connection.

I am grateful for any further tip!

pgrawehr, 2020-07-11:
Please try the development build. You shold find instructions on how to use it in the Readme or in the documentation folder.

krwq, 2020-07-21:
We have BME280 in our lab and it's tested on every build and CI run so I assume it works just fine. Please try latest and use code as following: https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio.Tests/ProtocolTests.cs#L55

Possibly there is some issue with the specific mode you're using - I don't have it right now to check

Ellerbach, 2020-09-14:
@D4rkF4ce, any news on this?

Ellerbach, 2020-09-23:
@D4rkF4ce, I will close this bug as I guess all the elements of answer are in this thread. Please feel free to reopen.

closed 2020-09-23 by Ellerbach