dotnet/iot #1135: RegisterCallbackForPinValueChangedEvent after restarting app not working on Orange pi zero plus (allwinner H5)

arnoud-koolecontrols, 2020-07-14:
I am a using a OrangePi zero plus board with a Allwinner H5.
Running Debian Buster
.NET Core SDK (3.1.106)
packages:
- iot.device.bindings 1.1.0-prerelease.20153.1
- system.device.gpio 1.1.0-prerelease.20153.1

In my application I use 
`
IoController = new GpioController();
if (this.Pinning.Enc0 > -1)
{
	this.IoController.OpenPin(this.Pinning.Enc0, PinMode.Input);
	this.IoController.RegisterCallbackForPinValueChangedEvent(this.Pinning.Enc0, PinEventTypes.Falling, Enc0Changed);
}
`
In the callback I write a text to the console. Nothing fancy just testing..
If I boot the application for the first time after booting the board, everything seems to work fine.
After restarting the application (CTRL-C and then sudo dotnet run) the callback does not work anymore.

Please let me now if I can do some kind of a test or should change my init.

pgrawehr, 2020-07-14:
What do you do to trigger the interrupt? With the given setting, only falling edges will fire the interrupt. 

I can't think of anything that should be different when starting the application a second time.

arnoud-koolecontrols, 2020-07-14:
I have connected a pullup resistor of 10k to 3,3V.
I have also attached a button which switches to ground.
I switch I couple of times to look at the console output.

It is very strange. If I close the application and start it again the event is not triggered anymore. 
To get it working again I have to reboot the board. 

Perhaps I can try to output the input every second. If this also stops working after restarting the app, then it looks like something is not freeing the GPIO claim properly.

arnoud-koolecontrols, 2020-07-14:
Ok I have just tested with the polling and I noticed that after the first poll the events are starting to work again.

The resulted in the following working workaround:
```
IoController = new GpioController();
if (this.Pinning.Enc0 > -1)
{
	this.IoController.OpenPin(this.Pinning.Enc0, PinMode.Input);
	this.IoController.RegisterCallbackForPinValueChangedEvent(this.Pinning.Enc0, PinEventTypes.Falling, Enc0Changed);
	this.IoController.Read(this.Pinning.Enc0);
}
```

Only logical explanation I can think of is that perhaps reading the GPIO resets the IRQ flag register.
Should I close this issue or should I keep it open?

pgrawehr, 2020-07-15:
Which Version of the Library are you using? Maybe test with the latest master build, because there where quite a few fixes in that area. If it reproduces there, this should clearly be fixed.

arnoud-koolecontrols, 2020-07-15:
Ok I updated to the latest version I can find: 1.1.0-prerelease.20276.1
In this version everything is working as expected.

So this seems to be fixed.

closed 2020-07-15 by arnoud-koolecontrols