dotnet/iot #1139: PN5180 ListenToCardIso14443TypeA hangup

arnoud-koolecontrols, 2020-07-23:
Don't know if this is the place to post problems with device bindings.
Please let me know if it should be posted somewhere else.

**PN5180 ListenToCardIso14443TypeA hangup**
I am looking into the device binding with the PN5180 to read and write some Mifare 1k classic cards.
I noticed that when a Mifare classic card is put in and out of the field the PN5180 stops working and the Busy line of the PN5180 stays high. 

**Steps to reproduce**
Call the ListenToCardIso14443TypeA  in a loop and hold a Mifare classic 1k card with a UID4 in front of the reader and away again. Repeat this until the hang-up occurs. 

**Possible fix**
at line 1036 in Pn5180.cs the anticollision is done. 
```csharp
for (int i = 0; i < 3; i++)
{
    // Select SL1
    uidSak[0] = (byte)(0x93 + i * 2);
    // NVB = Number of valid bits
    uidSak[1] = 0x20;
    SendDataToCard(uidSak.Slice(0, 2));
    // Read 5 bytes sak. Byte 1 will tell us if we have the full UID or if we need to read more
    ReadDataFromCard(sakInterm.Slice(0, 5));
```
To prevent the hang-up a check if 5 bytes are received would be enough.
```csharp
for (int i = 0; i < 3; i++)
{
    // Select SL1
    uidSak[0] = (byte)(0x93 + i * 2);
    // NVB = Number of valid bits
    uidSak[1] = 0x20;
    SendDataToCard(uidSak.Slice(0, 2));
    (numBytes, _) = GetNumberOfBytesReceivedAndValidBits();
    if (numBytes != 5)
    {
        return false;   //this will prevent a hangup
    }
    // Read 5 bytes sak. Byte 1 will tell us if we have the full UID or if we need to read more
    ReadDataFromCard(sakInterm.Slice(0, 5));
```

krwq, 2020-07-24:
cc: @Ellerbach

arnoud-koolecontrols, 2020-07-24:
After fixing this hang-up I also notice the ListenToCardIso14443TypeA stops working after reading some blockdata of a mifare classic 1k card.

After authenticating for a block the MFC_CRYPTO_ON bit in the SYSTEM_CONFIG is set high. 

**Possible fix**
at line 1022 in Pn5180.cs the flags are configured

```csharp
// Clears all interrupt
SpiWriteRegister(Command.WRITE_REGISTER, Register.IRQ_CLEAR, new byte[] { 0xFF, 0xFF, 0x0F, 0x00 });
// Sets the PN5180 into IDLE state
SpiWriteRegister(Command.WRITE_REGISTER_AND_MASK, Register.SYSTEM_CONFIG, new byte[] { 0xF8, 0xFF, 0xFF, 0xFF });
// Activates TRANSCEIVE routine
SpiWriteRegister(Command.WRITE_REGISTER_OR_MASK, Register.SYSTEM_CONFIG, new byte[] { 0x03, 0x00, 0x00, 0x00 });
// Sends REQB command
```
we should also reset the MFC_CRYPTO_ON flag
```csharp
// Clears all interrupt
SpiWriteRegister(Command.WRITE_REGISTER, Register.IRQ_CLEAR, new byte[] { 0xFF, 0xFF, 0x0F, 0x00 });
// Sets the PN5180 into IDLE state
SpiWriteRegister(Command.WRITE_REGISTER_AND_MASK, Register.SYSTEM_CONFIG, new byte[] { 0xB8, 0xFF, 0xFF, 0xFF });
// Activates TRANSCEIVE routine
SpiWriteRegister(Command.WRITE_REGISTER_OR_MASK, Register.SYSTEM_CONFIG, new byte[] { 0x03, 0x00, 0x00, 0x00 });
// Sends REQB command
```

Ellerbach, 2020-07-27:
Thanks @arnoud-koolecontrols , I will look at this as soon as I have access to the hardware to realize a test.
Have you tried the fix yourself? Is it fixing the issue or you still have it after some time?

arnoud-koolecontrols, 2020-07-27:
@Ellerbach I am using a copy of the binding and renamed the namesspace. This made it possible for me to experiment and check the behavior in the original version and the local version. I have used it now for a while and have not yett noticed any hangs anymore, while reading mifare classic 1k cards or connecting to my phone. 
So I am very pleased with the available binding. :-)

Ellerbach, 2020-07-27:
@arnoud-koolecontrols you can even then propose a PR to correct this issue. That may be much faster than me having access to my hardware to do it and test it. 
Thanks again for the feedback and for finding a fix for this bug.

krwq, 2020-08-06:
@arnoud-koolecontrols is this issue already fixed by your PR? Please close it if so... (note: you can put something like `Fixes #1139` in first line of PR description, that will auto-close the issue when PR is merged)

closed 2020-08-11 by arnoud-koolecontrols