dotnet/iot #1140: MPU9250.sample.cs After running calibration sowing wrong values

VincentvanWijnbergen, 2020-07-25:
When i run the sample script MPU9250.sample.cs I get wrong values for the accelerometer and gyroscope.
This only happens when the CalibrateGyroscopeAccelerometer() is used.

If I try to get the values without calibaration everything works as expected.

[image]

[image]

krwq, 2020-07-27:
@VincentvanWijnbergen, sanity check: are you rotating your IMU in all directions while calibrating? (rotate in all 3 axes) Otherwise it will likely not calibrate correctly

Oostwegel, 2020-07-30:
> @VincentvanWijnbergen, sanity check: are you rotating your IMU in all directions while calibrating? (rotate in all 3 axes) Otherwise it will likely not calibrate correctly

I have the same problem. I turned the IMU in all directions. Retried several times. Sometimes it gives the static values ​​as in the picture. And other times it gives values ​​that are completely wrong. I initially thought there was something wrong with the chip so I tried it with python and it worked perfectly. [python lib used](https://pypi.org/project/mpu9250-jmdev/)

So why is this not working in .net ?

krwq, 2020-08-06:
@Oostwegel note that Iot.Device.Bindings is community supported package and this sensor is relatively new and I suspect everything worked for person developing it and they likely didn't notice the issue.

I don't have this sensor or enough time to help you but I suspect good starting point would be to compare:
https://github.com/dotnet/iot/blob/master/src/devices/Mpu9250/Mpu6500.cs#L526
with
https://github.com/Intelligent-Vehicle-Perception/MPU-9250-Sensors-Data-Collect/blob/master/mpu9250_jmdev/mpu_9250.py#L610

and perhaps shortcut path might be to see if there is some relationship between value returned and value expected and perhaps it is related to some of these:
https://github.com/dotnet/iot/blob/master/src/devices/Mpu9250/samples/Mpu9250.sample.cs#L123-L128
as the error might be just using `+` instead of `-` in one of the formulas

Ellerbach, 2020-08-21:
> So why is this not working in .net ?

There must be an issue then in the calculation for the calibration. I originally wrote the code and remember it worked. But if it's not working for you, then it means something is wrong. I won't have a physical access to the sensor before at least 3 weeks. I will look anyway in the doc and the code in the mean time to find out why your calibration is not working.

@VincentvanWijnbergen  and @Oostwegel , would be great if you can share the results with and without the calibration. Also make sure you perform a hard reset of the sensor before the first calibration. And if you run it a second time, send the results as well. I suspect the code is not resetting the previous saved data and this can impact then the calibration.

Ellerbach, 2020-12-23:
Closing this one as it's fixed now.

closed 2020-12-23 by Ellerbach