dotnet/iot #1141: Suggest writing I2C routines to return success/error value so that programmer can choose output message

tardisgallifrey, 2020-07-26:
I'm using the I2C connection routines and Write method in System.GPIO to drive an MCP4725 DAC board.  It is on a Raspberry Pi 4, running raspbian Buster with dotnet core 3.1 installed.

I have it working, that alone is not the issue.  However, if I do something such as forget to run with 'sudo' (the most common error) or it can't reach the device on I2C, I would prefer the ability to control what occurs during an error.

I've tried enclosing:
            I2cConnectionSettings settings = new I2cConnectionSettings(1, 0x60);
            I2cDevice mcp4725 = I2cDevice.Create(settings);

In a try/catch block, but what occurs is that the two lines above don't run.  Therefore, when I call an mcp4725.Write() later, it fails.

I've also noticed that the methods like Write() are all void and do not return values.  It would be helpful if they did so that the programmer could control error actions/events.

If I were in C, which I have also written the same code, we use file descriptors and open().  If the file descriptor is null, I can choose what to say in the error message.  Usually, the first message is, "DID YOU RUN THIS AS SUDO!"

The Create constructor above does return a string, but it returns the same string whether it succeeds for fails.

Because I eventually wish to place most of the setup and action code in a class, and I plan to have each method return an error number for a main program to take action upon, I'd rather have something like this:

            I2cConnectionSettings settings = new I2cConnectionSettings(1, 0x60);
            int myError = I2cDevice mcp4725 = I2cDevice.Create(settings);
            if(myError != -1){
                      //do normal stuff
            }else{

                     Console.Writeline("An error occurred");
                     Console.Writeline("DID YOU RUN THIS AS SUDO!");
            }

I'm fairly new to C# and dotnet, so it is possible I'm missing something that is already there.  Try/Catch isn't working, but maybe something else would?

pgrawehr, 2020-07-27:
Please show us the code you run to generate an error. 
Neither of these lines
```
I2cConnectionSettings settings = new I2cConnectionSettings(1, 0x60);
I2cDevice mcp4725 = I2cDevice.Create(settings);
```
should cause an error, since they don't really talk to the device yet. That only happens after you call `mcp4725.Write()` or `mcp4725.Read()`. If something goes wrong there, you'll usually get an exception with error code 121 (device not responding). 

Note also that with an up-do-date Raspbian, sudo is _not_ required for I2C communication to work.

tardisgallifrey, 2020-07-27:
My Raspbian is at Buster and updated.  It still requires 'sudo' to run GPIO tasks.  As far as I can tell, I can't upgrade to a new distribution without re-imaging the SD card, which I'm not ready to do.  

My code is just a test program to see how things work with the MCP4725.  The code I've included does run correctly and produces a voltage ramp on the output of the board.  I have tried wrapping the two initialization lines in a try/catch block. Yet, that I've already explained.

I've also wrapped the mcp4725.Write() in a try/catch block; yet, I don't think it acted as I'd expected.  That is also when I got a message from the compiler that Write() doesn't return a value because it is a 'void' type.

To restate what you put into your reply: the initialization lines won't generate an error because they aren't actually talking to the I2C bus; therefore, in order to use a try/catch block, I'd need to use a try/catch block on a Write() method.  I'd need to send a Write() even if I just want to check to see if the bus is connected and board is active, correct?

My code is as close to my original C code as I could manage.  It increases output by 100 until reaching max of 4096 (0 to max VCC).

Current code is:
using System;
using System.Device.I2c;
using System.Diagnostics;
using System.Threading;
using System.Device.Gpio;

namespace MCP4725
{
    class Program
    {
        static void Main(string[] args)
        {

            int val;
            byte[] writeBuf = new byte[3];

        
            // set I2C bus ID: 1
        
            //Create a device with these settings: bus 1, address 0x60
            //This writes the first byte, I think
            I2cConnectionSettings settings = new I2cConnectionSettings(1, 0x60);
            I2cDevice mcp4725 = I2cDevice.Create(settings);

                    
            //control byte
            //0x40 = normal
            //0x60 = write EEPROM
            writeBuf[0] = 0x40;

            //These 2 bytes make write register
            //Just setting up, using later
            writeBuf[1] = 0;        //MSB
            writeBuf[2] = 0;        //LSB

            int count = 0;
            while(count < 4096){

                val = count;

                //val is an int (16bits)
                //writeBuf is a byte (8bits)
                //this method lops off the extra bits
                //so that an overflow error isn't produced
                writeBuf[1] = (byte)(val >> 4);
                writeBuf[2] = (byte)(val << 4);

                mcp4725.Write(writeBuf);
                Console.WriteLine($"{val}     {writeBuf[1]}     {writeBuf[2]}");
                count += 100;
                Thread.Sleep(2000);
            }
        }
    }    
}

pgrawehr, 2020-07-28:
> therefore, in order to use a try/catch block, I'd need to use a try/catch block on a Write() method. I'd need to send a Write() even if I just want to check to see if the bus is connected and board is active, correct?

Yes, that's correct. You can also use a Read() on any register, though (which normally has no side effect). That's actually what `i2cdetect` does: It sends a read command to every bus address and waits for a reply. If anything is listening on that device address, a value is read, if not, an exception is thrown.

tardisgallifrey, 2020-07-28:
I will try this on Saturday when next I work on the Pi.  I'll let y'all know if it works as expected.

Thanks,

tardisgallifrey, 2020-08-01:
Good Morning,

As promised, I worked on the MDP4725 C# program again today.  I used a try/catch block on an mcp4725.Read() as a check on whether there was access to the I2C bus and the device.

It worked.

I also wrapped the mcp4725.Write() in a try/catch block and it worked.

It took a couple of runs to try and get everything to work correctly, but it is suppressing the overflowing error message and it writing what I tell it from the correct try/catch block.  I mean no offense to y'alls error messages, but I prefer to tell the user something more concrete.  I'm writing towards regular users that may not understand that type of message.

Thanks for your help.  

If interested, this project is kept at: https://github.com/tardisgallifrey/mcp4725

The new code for this test program is:

     using System;
     using System.Device.I2c;
     using System.Diagnostics;
     using System.Threading;
     using System.Device.Gpio;

     namespace MCP4725
     {
         class Program
         {
             static void Main(string[] args)
             {

                int val;
                byte[] writeBuf = new byte[3];

                 // set I2C bus ID: 1

                 //Create a device with these settings: bus 1, address 0x60
                 //This writes the first byte, I think
                 I2cConnectionSettings settings = new I2cConnectionSettings(1, 0x60);
                 I2cDevice mcp4725 = I2cDevice.Create(settings);

                 //control byte
                 //0x40 = normal
                 //0x60 = write EEPROM
                 writeBuf[0] = 0x40;

                 //These 2 bytes make write register
                 //Just setting up, using later
                 writeBuf[1] = 0;        //MSB
                 writeBuf[2] = 0;        //LSB

                 //try to read from address
                 /If it fails, first check
                //if you ran as SUDO
                try
                
                     mcp4725.Read(writeBuf);
                 }
                catch(System.IO.IOException e)
                {
                     Console.WriteLine("DID YOU RUN THIS AS SUDO!");
                     Console.WriteLine("Possible I2C bus error");
                     Console.WriteLine("Check wiring of SDA and SCL");
                     System.Environment.Exit(0);

                 }

                 int count = 0;
                 while(count < 4096){
                     val = count;

                     //val is an int (16bits)
                     //writeBuf is a byte (8bits)
                    //this method lops off the extra bits
                     writeBuf[1] = (byte)(val >> 4);
                     writeBuf[2] = (byte)(val << 4);

                     //try to write to the registers
                     //if it fails, print message
                    try
                    {

                         mcp4725.Write(writeBuf);

                     }
                     catch(System.IO.IOException e)
                     {
                         Console.WriteLine("Error occurred during last write.");
                         Console.WriteLine("Possible problem with I2C bus or MCP4725 board");
                         Console.WriteLine("Exiting.  Sorry.");
                         System.Environment.Exit(0);
                
                     }

                         Console.WriteLine($"{val}     {writeBuf[1]}     {writeBuf[2]}");
                         count += 100;
                        Thread.Sleep(2000);
                 }
             }
         }
     }

closed 2020-08-01 by tardisgallifrey