dotnet/iot #1153: Dht22 failed to read. I can read it using Python to make sure that the connection is OK.

bbhxwl, 2020-08-11:
Dht22 failed to read. I can read it using Python to make sure that the connection is OK.

 
using (Iot.Device.DHTxx.Dht22 dht = new Iot.Device.DHTxx.Dht22(26, PinNumberingScheme.Logical))
                {
                    while (true)
                    {
                        Console.WriteLine($"Temperature: {dht.Temperature.Celsius.ToString("0.0")} ℃, Humidity: { dht.Humidity} %");
                        Thread.Sleep(1000);
                    }
                }

pgrawehr, 2020-08-12:
What is rhe error you get? I have no Dht22, but I know the Dht11 works fine.

bbhxwl, 2020-08-12:
> What is rhe error you get? I have no Dht22, but I know the Dht11 works fine.

I have also tested DHT11, but I can't get any data. Nan, I don't know why.

Ellerbach, 2020-08-21:
Which hardware platform are you using? Which OS? 
DHT sensors are very sensitive, depending on the hardware/OS, the managed driver may not work perfectly.

bbhxwl, 2020-08-23:
> I tested with creating the dht device in the following ways:
> 
> ```
> var dht = new Dht11(4, gpioController: new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver()));
> var dht = new Dht11(4);
> ```
> 
> In my case both worked the same. I then made some code to read the temperature, this code may need to be called several times until dht.IsLastReadSuccessful is true, with some delay between each call. It could take 5-10 seconds maybe before it was stable, but I did'nt really time it.
> 
> ```
> try
> {
> 	var temp = dht.Temperature;
> }
> catch(ArgumentException)
> {
>  // ignore
> }
> ```
> 
> When it was up and running I used something like this
> 
> ```
> if (dht.IsLastReadSuccessful)
> {
> 	var temperature = dht.Temperature.DegreesCelsius;
> 	var humidity = dht.Humidity;
> }
> ```
> 
> But the get properties should probably always be wrapped in a try/catch as the IsLastReadSuccessful only account for the last call not the new one made by the call to the Temperature property. I haven't used this code for anyting serious, just some fun programming exercise with my nephew on the CrowPi.

It seems that the current bug is not suitable for development.

bbhxwl, 2020-08-23:
> I tested with creating the dht device in the following ways:
>
> ```
> var dht = new Dht11(4, gpioController: new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver()));
> var dht = new [...]

        using (Iot.Device.DHTxx.Dht11 dht = new Iot.Device.DHTxx.Dht11(26, PinNumberingScheme.Logical))
            {
                while (true)
                {
                    var vv = dht.Temperature;
                    var vvv = dht.Humidity;
                    if (dht.IsLastReadSuccessful)
                    {
                        Console.WriteLine($"Temperature: {vv.Celsius.ToString("0.0")} ℃, Humidity: { vvv} %");
                        Thread.Sleep(1000);
                    }
                    
                }
            }

[image]

bbhxwl, 2020-08-23:
Dht22 acquisition often fails, but when the temperature is obtained, it is also wrong. Why?

Temperature: NaN ℃, Humidity: 76.3 %
Temperature: 3.6 ℃, Humidity: 76.3 %
Temperature: 3.6 ℃, Humidity: 76.3 %
Temperature: NaN ℃, Humidity: 74.9 %
Temperature: 3.6 ℃, Humidity: 76.4 %
Temperature: NaN ℃, Humidity: 75.9 %
Temperature: 3.5 ℃, Humidity: 76.3 %
Temperature: 3.6 ℃, Humidity: 74.9 %
Temperature: 3.6 ℃, Humidity: 75 %
Temperature: 3.6 ℃, Humidity: 76.5 %
Temperature: 3.5 ℃, Humidity: 76.10000000000001 %
Temperature: NaN ℃, Humidity: 76.2 %
Temperature: 3.5 ℃, Humidity: 76.80000000000001 %
Temperature: 3.5 ℃, Humidity: 76.80000000000001 %
Temperature: 3.4 ℃, Humidity: 76.3 %
Temperature: 3.4 ℃, Humidity: 75.2 %
Temperature: NaN ℃, Humidity: 76.3 %
Temperature: 3.3 ℃, Humidity: 75.2 %
Temperature: 3.3 ℃, Humidity: 76.3 %
Temperature: 3.3 ℃, Humidity: 76 %
Temperature: NaN ℃, Humidity: 76.60000000000001 %

Ellerbach, 2020-08-24:
> Dht22 acquisition often fails, but when the temperature is obtained, it is also wrong. Why?

Seems like the calculation for DHT22 is wrong. Should be more like:
```csharp
var temp = (((readBuff[2] & 0x7F) << 8) + readBuff[3] )* 0.1;
if (readBuff[2] & 0x80 == 0x80)
{
    temp = -temp;
}
```csharp

Can anyone try? Not sure if I have the hardware to test it out.

PR #1164 (2020-09-15): Fixing DHT issues, adding and adjusting documentation
Fixing DHT sensor issues:
- Fixes #1153, fixes #1161, should fix #1112 
- Making consistent calculation for temperatures for all DHT sensors
- Fixing DHT21 temperature calculation
Adjusting the documentation:
- Adding more sample for GPIO 1 wire usage
- Adding documentation for propre usage of DHT sensor
- Adding FAQ
Adding more sample:
- Sample implementing both I2C and 1 wire
- Ability to choose the sensor and the pin
closed 2020-09-15 by Ellerbach