dotnet/iot #1173: Adding Device-Binding Template to New VS Project Dialog

shaggygi, 2020-09-03:
What needs to be updated in order for the device binding template to show up in the new VS Project Dialog that is previewed in VS 16.8P2?  I attempted to verify by installing locally again, but doesn't appear to show.  I tried searching by textbox search and using the dropdowns (i.e. IoT, which is one of the categories we originally added in template.json).

Related content:
https://github.com/dotnet/iot/tree/master/tools/templates/DeviceBindingTemplate
https://devblogs.microsoft.com/dotnet/net-cli-templates-in-visual-studio/

sayedihashimi, 2020-09-09:
The issue here has nothing to do with the template itself. The issue is that currently Visual Studio we do not support showing templates that have been installed via a folder path.

@shaggygi can you file a new developer community Suggestion, and then share the link with me? We can use that to track the status of getting this new feature added.

shaggygi, 2020-09-09:
@sayedihashimi I just created suggestion.

https://developercommunity.visualstudio.com/content/idea/1179562/show-templates-installed-by-folder-path-in-vs-new.html

Thanks for reaching out.  It will be helpful.

sayedihashimi, 2020-09-09:
@shaggygi thanks for the link, I have moved it to my team's area. Let's see how much interest we get from the community on that issue. The more interest, the easier it will be for us to get it prioritized.

Ellerbach, 2021-07-01:
[Triage] Blocked as waiting for the Visual Studio team to answer the suggestion.

sayedihashimi, 2021-07-06:
As I mentioned in my [previous comment](https://github.com/dotnet/iot/issues/1173#issuecomment-689801063), Visual Studio doesn't support showing templates from a folder path. The template must be in a nuget package. In addition to this, I briefly looked at the template, and it isn't setup to properly support Visual Studio. You'll need to add an icon at least for VS support, but you may want to tweak additional things as well. I have some guidance over at https://github.com/sayedihashimi/template-sample on how to prep the templates.
After you have the template in a nuget package, it would be good to publish that to nuget.org so that users don't need to fork the repo to install the template.

Regarding building the template into Visual Studio, we are trying to simplify the built in templates and reduce the list of preinstalled templates. IMO it would be better to get this template to support Visual Studio and have users install from nuget.org for now. If the template becomes popular, or if there are a lot of requests to build it in (currently the score on the devcom issue is low), then we should consider the template to be built in.

cc @timheuer

krwq, 2021-07-07:
marking as untriaged so that it shows again on our triage radar

phenning, 2022-05-24:
Note that in Visual Studio 2022, we do support template engine base templates being used in the new project dialog via dotnet new --install [path]

shaggygi, 2022-05-27:
Does appear to now work in VS17.  Use the following to add...

```dotnet new -i C:\[PATH]\iot\tools\templates\DeviceBindingTemplate\dotnet_new_device-binding_csharp```

closed 2022-05-27 by shaggygi

phenning, 2022-05-27:
@sayedihashimi , @shaggygi  FYI, I resolved the Suggestion ticket as well, since we've had this since 17.0