dotnet/iot #1177: GPIO Raspberry PI Compute Module 3

bPoller2810, 2020-09-09:
I found the https://github.com/dotnet/iot/issues/762 but its pretty silent in there. 
Also the https://github.com/dotnet/iot/pull/1034 is kinda dead.

Is there someone here who knows about the progress we can expect there?

in short we want to just be able to use the following 3 GPIO:
- 30 Read
- 37 Write
- 31 Write

pgrawehr, 2020-09-09:
Unfortunately, the responsible people for final review of the open PRs are very busy recently. I too hope this gets better soon.

bPoller2810, 2020-09-10:
Well the whole problem was reported 11 month ago (ish) and seemed fixed 5 month ago. 

maybe i need too look if there is other projects out there, if it issnt certain that problems will be fixed

krwq, 2020-10-02:
I believe this should be fixed with @pgrawehr's PR above

krwq, 2020-10-02:
(and sorry for not being able to take a look at the PRs sooner, I believe it should improve now)

closed 2020-10-02 by krwq