dotnet/iot #1181: WeatherHelper Pressure functions should return UnitsNet

Ellerbach, 2020-09-14:
**Describe the bug**

In the WeatherHelpers, all the Pressure functions should return UnitsNet. The ```CalculateAltitude``` are returning double while they should return Length

**Steps to reproduce**

see here: https://github.com/dotnet/iot/blob/3c4034781d158b26be0bc4866c34192d86bc4ef1/src/devices/Common/Iot/Device/Common/WeatherHelper.cs#L159

**Expected behavior**

Return a Length and not a double

**Versions used**

Source code from latest master baunch.

cshung, 2020-09-15:
Hi maintainers - I would like to give this a try.

joperezr, 2020-09-15:
Sounds great! let us know if you need any help building the repo or submiting your contribution, we are happy to help 😀

krwq, 2020-09-24:
@cshung not sure if you got anywhere with this, I think @pgrawehr has addressed this as part of other issue

cshung, 2020-09-24:
I was planning to do this in October as part of Hacktoberfest. 
If it has to be done before October or it is already done somehow then I can let this go, no worries.

pgrawehr, 2020-09-24:
This will be fixed with #1178.

Ellerbach, 2020-09-25:
As it's now fixed by @pgrawehr , closing this issue! 
@cshung we do have quite some other issues open :-D you're welcome to pick another one :-D

closed 2020-09-25 by Ellerbach