dotnet/iot #1190: Move samples projects to .NET 5.0

richlander, 2020-09-21:
We should decide when we can do this. Can we do it now? Wait until .NET 5.0 RTM?

It would be really nice to move all of the samples to top-level programs, for example.

What do people think that use this repo? Do you use the samples (both at https://github.com/dotnet/iot/tree/master/samples and the ones that exist in samples directories at https://github.com/dotnet/iot/tree/master/src/devices)?

Ellerbach, 2020-09-22:
> Can we do it now?

You can always prepare everything...

> Wait until .NET 5.0 RTM?

...and push the PR once .NET 5.0 is final.

And if you update, better update everything for consistency. At least it will be easier.

shaggygi, 2020-09-22:
> It would be really nice to move all of the samples to top-level programs, for example.

Don't quite understand the comment.  Do you mean move all binding samples to https://github.com/dotnet/iot/tree/master/samples?  If so, I disagree. It seems to be better to keep the binding samples within respective binding folder.  That allows it to be discovered easier and updated when changes or new features occur.

As for https://github.com/dotnet/iot/tree/master/samples, this is a good location for newbies to get a quick sense of using APIs.

krwq, 2020-09-22:
I agree with @shaggygi that it's much easier to find when they are local to the project. What we can do is perhaps generate part of https://github.com/dotnet/iot/blob/master/samples/README.md to point to samples/READMEs from devices

pgrawehr, 2020-09-22:
@richlander : Due to #1099, the entire library currently is not working with .NET Core 5, preview 5 or later. So unless you use preview 4, you'll have a hard time converting anything to .NET 5.

richlander, 2020-10-25:
I wasn't proposing to move the samples from the binding folder. I also agree that we should not do that and they are best where they are.

@pgrawehr -- this is due to a breaking change w/rt to WinRT, related to https://github.com/dotnet/iot/issues/1193. 5.0 samples should now be resolved.

Once my open NRTs people gets merged, I'll create a PR for updating the samples.

closed 2020-10-28 by richlander