dotnet/iot #1199: DCMotor - compatibility with other drivers

wsad4ryba, 2020-09-27:
**Describe the bug**
I tried to use the DCMotor class to control the motor with a different controller than in the example.
When i passed small negative values to motor.speed the motor started spinning very fast.
My driver is MD13S:
https://www.cytron.io/p-13amp-6v-30v-dc-motor-driver
Wiring diagram:
[image]

**Steps to reproduce**

```csharp
DCMotor motor = DCMotor.Create(PwmChannel.Create(0, 1, 2000, 0.0), DIRpin, controler, false);

for (int i = 0; i < 360; i++)
{
    speedSet = Math.Sin(Math.PI * i / 180.0);
    System.Console.WriteLine($"{i}. Actual motor setting is: {speedSet}");
    motor.Speed = speedSet;
    Thread.Sleep(100);
}
```

**Expected behavior**
I start with small positive values, accelerate the engine to the maximum and then slow it down, after passing negative values it should behave the same way, but turn the other way.

**Actual behavior**
When small negative values appeared, the engine started to spin very quickly, then slowed down and then accelerated again, which is opposite to the positive values, and yet it should behave similarly but spin the other way.

I thought it was a simple bug with pwm.DutyCycle but it turned out to be a driver incompatibility, mine uses one directional pin, while the driver from the documentation example has two directional pins.
More information  in #1186 

**Versions used**
Version:   5.0.100-preview.6.20318.15
and cloned repository

krwq, 2020-09-29:
Thanks @wsad4ryba, as we talked in the PR, we should do as I commented here: https://github.com/dotnet/iot/pull/1186#issuecomment-698876501 so in short: rename argument in the existing API to not be confusing and not suggest that they are compatible with your driver and add new APIs to support your scenario (and make sure we cover that in the samples)

wsad4ryba, 2020-12-30:
PR is merged, so until some driver comes out that needs other control we can close this issue.

closed 2020-12-30 by wsad4ryba