dotnet/iot #1201: MCP23S17 Address Problem

Eroli, 2020-09-29:
Hello everyone,

I use two MCP23S17 in one circuit, where both chips run on the same ChipSelect line, but have different hardware addresses.

Chip A: Pin 15/16/17 on GND --> address 32
Chip B: Pin 15 to 3.3V via 10 kOhm resistor, Pin 16/17 to GND --> address 33

Nevertheless, when I address chip A via the library, both chips always react. The other way round, of course, this is also true. So I can't address the chips separately, so I guess that the address handling might contain an error?

Are there already any abnormalities in this direction? Does anyone use differently addressed MCP23S17 on the same ChipSelect line?

pgrawehr, 2020-09-29:
I haven't used the address lines with the MCP23S17, but reading the documentation, this needs setting the HAEN bit in the IOCON register (Register 0x5, Bit 3).

Eroli, 2020-09-30:
Dear @pgrawehr ,

indeed, this solved the issue. I have not been setting the HAEN bit. I have been struggling with the correct settings of the IOCON register, but now it works as expected. For other users: Please consider the correct addressing of the IOCON register, which is subject to change depending on the configuration. Also have a look at the errata!

Thank you very much for your help. Great service!

closed 2020-09-30 by Eroli