dotnet/iot #1203: Error on using ONLY SPI1 (Can not set SPI mode to Mode3)

wolfgangmauer, 2020-10-01:
Since I have a broken pin on SPI0, I have to use SPI1 on my **Raspberry Pi 3B+**
In config.txt i have set

```
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
dtoverlay=spi1-1cs,cs0_pin=12
```
pi@raspberrypi:~ $ ls -l /dev/spidev*
crw-rw---- 1 root spi 153, 0 Okt  1 11:36 /dev/spidev1.0
pi@raspberrypi:~ $

If i use this code:
```csharp
            var settings = new SpiConnectionSettings(1, 0)
            {
                ChipSelectLineActiveState = PinValue.Low,
                ClockFrequency = 2000000,
                DataBitLength = 8,
                DataFlow = DataFlow.MsbFirst,
                Mode = SpiMode.Mode3
            };
            using var spiDevice = SpiDevice.Create(settings);
            spiDevice.WriteByte((byte)(address | 0x80));
```
i get:
Unhandled exception. System.IO.IOException: Error 22. Can not set SPI mode to Mode3.
   at System.Device.Spi.UnixSpiDevice.Initialize()
   at System.Device.Spi.UnixSpiDevice.WriteByte(Byte value)

Add following information:

```
pi@raspberrypi:~ $ dotnet --info
.NET Core SDK (gemäß "global.json"):
 Version:   3.1.402
 Commit:    9b5de826fd

Laufzeitumgebung:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /opt/dotnet/sdk/3.1.402/

Host (useful for support):
  Version: 3.1.8
  Commit:  9c1330dedd

.NET Core SDKs installed:
  3.1.402 [/opt/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.8 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.8 [/opt/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
```
- Version of `System.Device.Gpio` package is 1.0.0
- Version of `Iot.Device.Bindings` package is 1.0.0

krwq, 2020-10-01:
@wolfgangmauer do other modes work or is it specific to Mode3? Can you try running as admin and seeing if still repros?

krwq, 2020-10-01:
Also can you check if https://github.com/nebrius/raspi-io/issues/52 also applies to you? Possibly this is similar issue and is just related to the config

krwq, 2020-10-01:
I think following might also apply to you, so there is a chance this is not supported:
https://www.raspberrypi.org/forums/viewtopic.php?t=149981

If the device is not super timing sensitive you can try using our SoftwareSpi, i.e. https://github.com/dotnet/iot/blob/master/src/devices/Mcp3xxx/samples/Mcp3008.Sample.cs#L22

wolfgangmauer, 2020-10-01:
@krwq THX a lot!
I'll try Software Spi, I hope my stepper (TMC2130) can still be controlled.

wolfgangmauer, 2020-10-01:
It would be fantastic if it were already implemented as a "device"

krwq, 2020-10-01:
@wolfgangmauer I'm not sure if this one is compatible with Uln2003: https://github.com/dotnet/iot/blob/master/src/devices/Uln2003/README.md

wolfgangmauer, 2020-10-01:
@krwq Unfortunately not compatible with Uln2003, the TCM2130 is configured via SPI

wolfgangmauer, 2020-10-04:
As far as I understand it, the bcm2835 has no way of setting extra settings for SPI1 (aux).
It seems that the settings are shared for both busses
I make a litte test with a working code
```
	if (bus == 0)
	{
		bcm2835_spi_begin();
		bcm2835_spi_setClockDivider(BCM2835_SPI_CLOCK_DIVIDER_256);
	}
	else
	{
		bcm2835_aux_spi_begin();
		bcm2835_aux_spi_setClockDivider(BCM2835_SPI_CLOCK_DIVIDER_256);
	}
	bcm2835_spi_setBitOrder(BCM2835_SPI_BIT_ORDER_MSBFIRST);
	bcm2835_spi_setDataMode(BCM2835_SPI_MODE3);
	bcm2835_spi_setChipSelectPolarity(BCM2835_SPI_CS0, LOW);
	bcm2835_spi_chipSelect(BCM2835_SPI_CS0);
```
So how can this be done with SpiDevice if i want to use SPI1 ?

krwq, 2020-10-05:
@wolfgangmauer we only use generic Linux implementation and go through linux APIs (ioctl): https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Spi/Devices/UnixSpiDevice.cs#L227 You would need to write RaspberryPi specific driver for SPI to bypass that and add extra settings on this specific board... As a short answer just create short interop layer to do exactly what you'd do in C...

If you find limitation on Raspberry Pi APIs you should file a bug in their repo to make sure they're at least aware of this. If they ack the issue and fix you should file another bug on Linux repo to make sure they update their code accordingly

Ellerbach, 2020-10-09:
@wolfgangmauer the question may look stupid but why do you want to use SPI1 and not SPI0? 
Even if you are already using SPI0, you may want to switch the devices? 
Another option is to use multiple Chip Select for SPI0 and have your steppers there? Maybe for the device using originally SPI0, the SoftwareSpi should work?
Would that help solving the issue?

wolfgangmauer, 2020-10-09:
@Ellerbach 
First post:
```
Since I have a broken pin on SPI0 ...
```
So far I haven't got the SoftwareSpi to run, I'll have to take a look at it with a logic analyzer.

wolfgangmauer, 2020-10-09:
@krwq 
```
we only use generic Linux implementation and go through linux APIs
```
Ok, I thought the System.Device.Gpio is Raspberry Pi specific :-(

krwq, 2020-10-12:
@wolfgangmauer it has lots of stuff Raspberry Pi specific but it's not. If possible we strive to it being board independent and with SPI/I2C this is possible. GPIO is a bit harder as it has higher perf expectations and generic implementation is relatively slow...

Ellerbach, 2021-07-01:
[Triage] As this seems to be Raspberry Pi related and based on the discussion, we will close this issue. Do not hesitate to reopen if needed.

closed 2021-07-01 by Ellerbach