dotnet/iot #1207: DCMotor - safe stop

wsad4ryba, 2020-10-12:
In #1186 I wanted to add ```Start()``` and ```Stop()``` to disable and enable pwm. 
Of course their effect would be the same as for the ```Speed = 0``` and ```Speed = <not a zero>```, but when the motor control would take place in separate threads, ```Speed = 0``` may not be sufficient. A good example would be the use of limit switches (my motor rotates the screw that moves the platform, which has limited movement, that is why I need limit switches). If we control the motor in a simple loop, e.g.
```csharp
for (int i = 0; i < 360; i++)
{
    speedSet = Math.Sin(Math.PI * i / 180.0);
    System.Console.WriteLine($"{i}. Actual motor setting is: {speedSet}");
    motor.Speed = speedSet;
    Thread.Sleep(100);
}
```
and use ```RegisterCallbackForPinValueChangedEvent``` for limit switch with ```Speed = 0``` will not break the loop so the motor will run again in the next iteration. You can check the condition of the limit switches in each control loop, but the use of events seemed safer to me. You only need to start monitoring events once and you don't have to worry about breaking something by accident.
I understand that checking the status of limit switches will have a similar effect, but you can accidentally forget about it when you try something new, and I just wanted to add additional protection, which of course is not obliged to use, as pwm would be enabled by default, and you can still only use Speed to control the engine.

krwq, 2020-10-12:
@wsad4ryba can this be implemented as a wrapper on top of DCMotor (written by hand, not tested):

```csharp
public class DCMotorWithStartStop : DCMotor
{
  private DCMotor _inner;
  private bool _stopped = false;
  private double _speed;

  public DCMotorWithStartStop(DCMotor innerMotor)
  {
    _inner = innerMotor;
    _speed = innerMotor.Speed;
  }

  public void Start()
  {
    _stopped = false;
    _inner.Speed = _speed;
  }

  public void Stop()
  {
    _stopped = true;
    _inner.Speed = 0.0;
  }

  public override double Speed
  {
    get => _speed;
    set
    {
      _speed = value;
      if (!_stopped)
      {
        _inner.Speed = _speed;
      }
    }
  }
}
```

wsad4ryba, 2020-10-13:
I tried but got error cs7036.
I was able to get rid of it thanks to these changes:
```csharp
        public DCMotorWithStartStop(GpioController controller, bool shouldDispose, DCMotor innerMotor)
            : base(controller, shouldDispose)
        {
            _inner = innerMotor;
            _speed = innerMotor.Speed;
        }
```
however, I don't like this solution because two parameters are passed twice:
```csharp
_motor = new DCMotorWithStartStop(_controller, false, DCMotor.Create(PwmChannel.Create(0, 1, 100, 0.0), motorDIRpin, _controller, false, true));
```
I am a beginner and I have not come up with anything better 😞 

Overall, the idea of the wrapper works very well, thank you very much 😄

krwq, 2020-10-13:
@wsad4ryba would `_inner.Controller` work instead of passing controller second time? `shouldDispose` feels ok, you should also have a default value in there

wsad4ryba, 2020-10-14:
I can't do it like that, the controller in DCMotor appears only in the protected constructor, or it is passed to internal classes with the create method, so I don't think I have access to it?
For now I have rearranged the order and added the default:
```csharp
public DCMotorWithStartStop(DCMotor innerMotor, GpioController controller, bool shouldDispose = false)
    : base(controller, shouldDispose)
{
    _inner = innerMotor;
    _speed = innerMotor.Speed;
}
```

krwq, 2020-10-15:
@wsad4ryba yes, you're right, perhaps we could add a new constructor to DCMotor to unblock this scenario, i.e.:

```csharp
public class DCMotor {
    protected DCMotor(DCMotor other) {
        Controller = other.Controller;
        // ... whatever else needs to be copied
    }
}
```

other option is to perhaps just make Controller public

wsad4ryba, 2020-10-18:
I experimented with this solution and it works even if nothing is copied.
In the DCMotor class I put this:
```csharp
protected DCMotor(DCMotor other)
        {
            // Controller = other.Controller;
            // _shouldDispose = other._shouldDispose;
        }
```
In the DCMotorWithStartStop class I put this:
```csharp
public DCMotorWithStartStop(DCMotor innerMotor)
            : base(innerMotor)
        {
            _inner = innerMotor;
            _speed = innerMotor.Speed;
        }
```
and in main program this:
```csharp
motor = new DCMotorWithStartStop(DCMotor.Create(PwmChannel.Create(0, 1, 100, 0.0), motorDirPin, controller, false, true));
```
I'm not sure if I understand correctly. Now DCMotorWithStartStop uses a new constructor from DCMotor which allows you to create a new object from an existing DCMotor object, so you don't have to copy anything because the whole object is copied by default?

krwq, 2020-10-19:
@wsad4ryba not what I had in mind and now understand the problem. What you can do is to `base(null, false)` instead to achieve the same effect but without adding new constructors. What you're gonna be missing with that is `DCMotorWithStartStop.Controller` will be always returning null which is not really bad considering it's protected and you can only access it from derived classes - what I suggested above is to fix that. Better solution perhaps might be to create parameterless constructor in DCMotor which does nothing

wsad4ryba, 2020-10-23:
I checked both scenarios, i.e. `base (null, false)` without an additional constructor in DCMotor and parameterless constructor in DCMotor without `base ()` in DCMotorWithStartStop. Both seem to be working.
Now the question is, which path do we want to continue and do you want this functionality in your repository?

wsad4ryba, 2020-12-30:
Sorry it's so late :)

PR #1398 (2021-01-11): Dc motor start stop
Fixes #1207
Adds a wrapper that allows the engine to stop or start, regardless of the Speed value.
closed 2021-01-11 by krwq