dotnet/iot #1210: CallbackForPinValueChangedEvent appears to unregister

sthompsonSymbrium, 2020-10-14:
**Describe the bug**

I am having a potential issue with the event I am registering with RegisterCallbackForPinValueChangedEvent stops firing
We are implementing a counter using the LibGpiodDriver on a RPi4, counting 8 channels of IO at a total freq of 20 Hz. (channels vary between  1Hz and 3Hz,  Each channel gets an event registered to it to maintain its count using PinEventTypes.Rising.  They will collect correctly initially, but after a while some channels will stop.

For example, on October 8 I started a test at 5:00 and here are the last times each channel collected 

Channel1    10/8/2020 17:21:27
Channel2    10/8/2020 22:35:50
Channel3    10/9/2020 10:16:10
Channel4    10/9/2020 5:36:37
Channel5    10/9/2020 10:16:10
Channel6    10/9/2020 10:16:10
Channel7    10/8/2020 22:19:52
Channel8    10/9/2020 10:16:10

Channels 3,5,6, and 8 are still firing, but the other 4 are not.  If I restart my application they will begin to fire again.

I started another test on October 13 and after an hour, 2 of the channels had stopped seeing the data pulses so I tried letting it rest overnight by stopping all of the data pulses but not restarting my application to see if the evens would re-register.  The next day I found that 4 more channels had died sometime in the night (I can't tell when because there were no pulses to count)  This makes me think that the channel dying is independent of the pulse stream coming into it.

I don't see any errors events to include

**Steps to reproduce**

```csharp
PiGPIO  gpiocontroller = new PiGPIO();

//for loop for 8 channels
{
                    //clear output
                    gpiocontroller.SetupPin(channel.ChannelIndex, false);
                    gpiocontroller.SetPinValue(channel.ChannelIndex, false);

                    if (channel.IsAnInput)
                    {
                        gpiocontroller.SetupPin(channel.ChannelIndex, true);
                        gpiocontroller.RegisterCallbackForPinValueChangedEvent(channel.ChannelIndex, PinEventTypes.Rising, PinChangeEventHandler);
                    }
}
// end for loop

//meanwhile in my PiGPIO gpiocontroller

GpioController gpiocontroller = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver());

public void SetupPin(int channelIndex, bool isInput)
        {
            if (gpiocontroller != null)
            {
                if (isInput)
                {
                    gpiocontroller.OpenPin(inputPinIndex[channelIndex], PinMode.Input);
                }
                else
                {
                    gpiocontroller.OpenPin(outputPinIndex[channelIndex], PinMode.Output);
                }
            }
        }

public void SetPinValue(int channelIndex, bool value)
        {
            if (gpiocontroller != null)
            {
                if (value)
                {
                    log.DebugFormat("Setting Channel {0} on Pin {1} to {2}", channelIndex, outputPinIndex[channelIndex], value);
                    gpiocontroller.Write(outputPinIndex[channelIndex], PinValue.Low);
                }
                else
                {
                    gpiocontroller.Write(outputPinIndex[channelIndex], PinValue.High);
                }
            }
        }

   public void RegisterCallbackForPinValueChangedEvent(int channelIndex, PinEventTypes rising, PinChangeEventHandler pinChangeEventHandler)
        {
            gpiocontroller.RegisterCallbackForPinValueChangedEvent(inputPinIndex[channelIndex], PinEventTypes.Rising, pinChangeEventHandler);
        }
//

```

**Expected behavior**

After 8 hours all 8 channels should be able to detect and trigger a PinChangeEvent

**Actual behavior**

After 8 hours (or infinite hour really) all 8 channels some of the channels can no longer detect and trigger a PinChangeEvent

**Versions used**

Add following information:

- `dotnet --info` on the machine being used to build
.NET Core SDK (reflecting any global.json):
 Version:   3.1.402
 Commit:    9b5de826fd

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18363
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.1.402\

Host (useful for support):
  Version: 3.1.8
  Commit:  9c1330dedd

.NET Core SDKs installed:
  2.1.202 [C:\Program Files\dotnet\sdk]
  2.1.505 [C:\Program Files\dotnet\sdk]
  2.1.507 [C:\Program Files\dotnet\sdk]
  2.1.509 [C:\Program Files\dotnet\sdk]
  2.1.602 [C:\Program Files\dotnet\sdk]
  2.1.801 [C:\Program Files\dotnet\sdk]
  2.2.402 [C:\Program Files\dotnet\sdk]
  3.1.402 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.0.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.22 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.8 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download

- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
self contained

- Version of `System.Device.Gpio` package
1.1.0-prerelease.20276.1

- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)
not sure

- Version of libgpiod-dev - "apt list libgpiod-dev"
libgpiod-dev/stable,now 1.2-3+rpi1 armhf [installed]

- Version of Linux - "cat /etc/os-release"
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

- Hardware info -  "cat /proc/cpuinfo"
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02100
Serial          : 00000000beb17867
Model           : Raspberry Pi Compute Module 3 Plus Rev 1.0

sthompsonSymbrium, 2020-10-30:
New information:
I had an idea to try once an hour to call UnregisterCallbackForPinValueChangedEvent then RegisterCallbackForPinValueChangedEvent for each of my channels of IO.  I started a test doing that yesterday and 7 channels are still collecting today 24 hours later, but one channel has not collected a pulse for the last 15 hours.  So, that means that my idea didn't work.

Ellerbach, 2020-10-31:
@sthompsonSymbrium , thanks for the info. I guess the model we are using with multithread and register to low level events is not the best for a large number of pin. It would require to rewrite this part to have only 1 thread checking the changes of all the pins at once. This is quite some work and need to be tested like you are doing. More investigation on our side needed. And thanks again for your findings.

Ellerbach, 2021-02-25:
[Triage] It seems that the way the event are handled would require to be redone to support subscription to multiple events at the same time rather than multiple subscriptions for single events. This is mainly located in the [event handler driver](https://github.com/dotnet/iot/blob/96fd9f1d4f334a43e0fd8a144cee1780eb0a0af5/src/System.Device.Gpio/System/Device/Gpio/LibgpiodDriverEventHandler.cs#L59).

Alex-111, 2021-03-11:
Are you sure the implementation is wrong ? Maybe the problem is just that an exception is happening in the long running loop?

PR #1499 (2021-05-06): ignore Interrupted system call error and retry
Fixes #1210
Libgpiod method gpiod_line_event_wait can be interrupted and return error code 4 (Interrupted system call).
In this case we can ignore Interrupted system call error continue waiting for events.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1499)
closed 2021-05-06 by joperezr