dotnet/iot #1216: Mcp23s17 cannot read registers

z-eh, 2020-10-18:
Hi!

After I patched the SpiAdapter.Read() function, to make it work, I realized that I cannot read any information from the registers.
Maybe I'm the fool, with my patch, but before the patch it throws ArgumentException, because the SpiAdapter.Read -> readBuffer and writeBuffer length not the same.

My patch in SpiAdapter.cs is at line 48:

From this:
```csharp
Span<byte> readBuffer = stackalloc byte[buffer.Length + 2];
```

To this:
```csharp
Span<byte> readBuffer = stackalloc byte[Math.Max(writeBuffer.Length, buffer.Length)];
```

I try to achive reading with GpioController class and with the Mcp23s17 class itself.

Thanks,
Zoltan

krwq, 2020-10-19:
Hey @z-eh can you share the error you're getting after applying that fix?

I suspect you'll need to calculate the size of the buffer (`Max(ReadBufferLength, WriteBufferLength)`) first and then use it for both write and read buffer and not just for the read buffer. Any reason you have closed https://github.com/dotnet/iot/pull/1208 ?

z-eh, 2020-10-19:
I closed it because I think this solution maybe the cause of my issue. When I make it work, then I made a new pull request with a working solution.

krwq, 2020-10-19:
sounds good, let me know if you need any help

Ellerbach, 2021-02-25:
[Triage] @z-eh did you manage to make it working? Does it need any patch inside the binding?

z-eh, 2021-02-25:
I can not manage to make it working...
It was a complete mess and I want to make sure to not brake anything else.
When 1.3 Release came out in December, then I tried it and it just failed again.
So I'm waiting new Raspberry Pi boards to arrive and new Mcp23s17 chips too. (+ a PiFace Digital I/O 2)
I want to make sure that it is not a hardware related issue on my side.
As I saw the code had "some" bugs in this area, so fingers crossed.
But my free work time for this is very limited.

closed 2022-09-30 by msftbot[bot]