dotnet/iot #1222: Unhandled exception trying to re-set PinMode with LibGpiodDriver

DanieleNeo, 2020-10-24:
**Bug description**

I was implementing DHT22 sensor and I got a strange error trying to read temperature and humidity values. The exception was thrown in the ReadThroughOneWire method, exactly when the method try to set the pin in InputPullup mode

``` 
internal virtual void ReadThroughOneWire()
        {
            byte readVal = 0;
            uint count;

            // keep data line HIGH
            _controller.SetPinMode(_pin, PinMode.Output);
            _controller.Write(_pin, PinValue.High);
            DelayHelper.DelayMilliseconds(20, true);

            // send trigger signal
            _controller.Write(_pin, PinValue.Low);
            // wait at least 18 milliseconds
            // here wait for 18 milliseconds will cause sensor initialization to fail
            DelayHelper.DelayMilliseconds(20, true);

            // pull up data line
            _controller.Write(_pin, PinValue.High);
            // wait 20 - 40 microseconds
            DelayHelper.DelayMicroseconds(30, true);

            _controller.SetPinMode(_pin, PinMode.InputPullUp);
```  

This is the exception

``` 
The pin does not support the selected mode.

   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)
   at ConsoleApp3.DhtBase.ReadThroughOneWire() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 162
   at ConsoleApp3.DhtBase.ReadData() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 130
   at ConsoleApp3.DhtBase.get_Temperature() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 68
   at ConsoleApp3.Program.Main(String[] args) in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\Program.cs:line 64
``` 
So I called the IsPinModeSupported method and I found that PinMode.InputPullUp is not supported (by LibGpiodDriver driver? I don't know...), not a big problem, my interface has hardware pullup, so I modified the PinMode.InputPullUp with PinMode.Input, discovering the more general issue

``` 
Error setting pin mode, for pin: 8, error: 16

   at System.Device.Gpio.Drivers.LibGpiodDriver.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)
   at ConsoleApp3.DhtBase.ReadThroughOneWire() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 164
   at ConsoleApp3.DhtBase.ReadData() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 130
   at ConsoleApp3.DhtBase.get_Temperature() in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\DhtBase.cs:line 68
   at ConsoleApp3.Program.Main(String[] args) in C:\Users\daniele\Source\Repos\ConsoleApp3\ConsoleApp3\Program.cs:line 64
``` 

Unfortunately this behavior is quite general and is caused by a second set of a pin already set previously, regardless of how it was previously set, using LibGpiodDriver. I use this driver because I'm developing a general solution running on OrangePI hardware, using Raspberry and its default driver issue does not happen.

**Steps to reproduce**

Using LibGpiodDriver is simple with this code

```
            var _driver = new System.Device.Gpio.Drivers.LibGpiodDriver();
            var _controller = new System.Device.Gpio.GpioController(System.Device.Gpio.PinNumberingScheme.Logical, _driver);
            var _pin = 8;
            _controller.OpenPin(_pin);
            int i = 0;

            while (true)
            {
                _controller.SetPinMode(_pin, PinMode.Output);

                var towrite = (++i % 2 > 0) ? PinValue.High : PinValue.Low;
                _controller.Write(_pin, towrite);
                Console.WriteLine("value written > " + towrite);

                _controller.SetPinMode(_pin, PinMode.Input);
                var valore = _controller.Read(_pin);
                Console.WriteLine("value read > " + valore);

                System.Threading.Thread.Sleep(1000);

            }
```

The code for the DHT22 sensor I can also rewrite it by handling PinMode.Input instead of PinMode.InputPullup but the fact that an already setted pin value cannot be changed I think it's a problem that needs to be solved.

Thanks in advance
D.

**`dotnet --info` on the machine being used to build**
```
.NET Core SDK (che rispecchia un qualsiasi file global.json):
 Version:   3.1.301
 Commit:    7feb845744

Ambiente di runtime:
 OS Name:     Windows
 OS Version:  10.0.19041
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.1.301\

Host (useful for support):
  Version: 3.1.5
  Commit:  65cd789777

.NET Core SDKs installed:
  3.1.301 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.19 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.19 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.5 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.19 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.5 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
```

**`dotnet --info` on the machine where app is being run (not applicable for self-contained apps)**
```
.NET Core SDK (reflecting any global.json):
 Version:   3.1.102
 Commit:    573d158fea

Runtime Environment:
 OS Name:     debian
 OS Version:  10
 OS Platform: Linux
 RID:         debian.10-arm
 Base Path:   /root/dotnet-arm32/sdk/3.1.102/

Host (useful for support):
  Version: 3.1.2
  Commit:  916b5cba26

.NET Core SDKs installed:
  3.1.102 [/root/dotnet-arm32/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.2 [/root/dotnet-arm32/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.2 [/root/dotnet-arm32/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
```

**Version of `System.Device.Gpio` package**
```
Stable 1.0.0 and 1.1.0-prerelease.20276.1
```

DanieleNeo, 2020-10-26:
Solution is #1226

D.

joperezr, 2021-04-15:
Sorry for taking long time to respond here. This will be fixed as part of https://github.com/dotnet/iot/pull/1473 by calling a realease to the lock on the line handle before changing the pinmode.

PR #1520 (2021-07-09): Opening/Closing a pin should not affect its state if possible
Fixes #1222 
Fixes #1479 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1520)
closed 2021-05-20 by msftbot[bot]