dotnet/iot #1226: OrangePiPCDriver

DanieleNeo, 2020-10-26:
I described my problem here #1222, I solved writing specific OrangePiPCDriver using the guidelines described here https://github.com/dotnet/iot/pull/780

This is the code that I ask to push in the repository

```
// Licensed to the .NET Foundation under one or more agreements.
// The .NET Foundation licenses this file to you under the MIT license.
// See the LICENSE file in the project root for more information.

namespace System.Device.Gpio.Drivers
{
    /// <summary>
    /// A GPIO driver for the Orange Pi PC.
    /// </summary>
    /// <remarks>
    /// SoC: Allwinner H3
    /// </remarks>
    public class OragePiPCDriver : SunxiDriver
    {
        protected override int GpioRegisterOffset0 => 0x01C20800;
        protected override int GpioRegisterOffset1 => 0x01F02C00;

        /// <summary>
        /// Orange Pi PC has 28 GPIO pins.
        /// </summary>
        protected internal override int PinCount => 28;

        /// <summary>
        /// Converts a board pin number to the driver's logical numbering scheme.
        /// </summary>
        /// <param name="pinNumber">The board pin number to convert.</param>
        /// <returns>The pin number in the driver's logical numbering scheme.</returns>
        protected internal override int ConvertPinNumberToLogicalNumberingScheme(int pinNumber)
        {
            return pinNumber switch
            {
                3 => MapPinNumber('A', 12),
                5 => MapPinNumber('A', 11),
                7 => MapPinNumber('A', 6),
                8 => MapPinNumber('A', 13),
                10 => MapPinNumber('A', 14),
                11 => MapPinNumber('A', 1),
                12 => MapPinNumber('C', 14),
                13 => MapPinNumber('A', 0),
                15 => MapPinNumber('A', 3),
                16 => MapPinNumber('C', 4),
                18 => MapPinNumber('C', 7),
                19 => MapPinNumber('C', 0),
                21 => MapPinNumber('C', 1),
                22 => MapPinNumber('A', 2),
                23 => MapPinNumber('C', 2),
                24 => MapPinNumber('C', 3),
                26 => MapPinNumber('A', 21),
                27 => MapPinNumber('A', 19),
                28 => MapPinNumber('A', 18),
                29 => MapPinNumber('A', 7),
                31 => MapPinNumber('A', 8),
                32 => MapPinNumber('E', 8),
                33 => MapPinNumber('A', 9),
                35 => MapPinNumber('A', 10),
                36 => MapPinNumber('E', 9),
                37 => MapPinNumber('A', 20),
                38 => MapPinNumber('E', 6),
                39 => MapPinNumber('E', 7),
                _ => throw new ArgumentException($"Board (header) pin {pinNumber} is not a GPIO pin on the {GetType().Name} device.", nameof(pinNumber))
            };
        }
    }
}
```

Thanks in advance,
D.

Ellerbach, 2020-10-27:
Thanks and good you found the issue! For the name of the calls, I guess `OrangePiPcDriver` shoud be a better naming.
In terms of namespace and place for the driver, if you can make an automatic detection, I guess we can discuss it going to System.Device.Gpio.Driver, if not, this should go to IoT.Device.Bindings

ZhangGaoxing, 2020-10-29:
I don't think .NET IoT will support Orange Pi in the near future because there are some problems in the process of submitting the driver. You can take a look at [this](https://github.com/ZhangGaoxing/sunxi-gpio-driver) first. If possible, you can help me test it on the Orange Pi PC, because I don't have this board. If you have any problems, you can tell me.😄

krwq, 2020-10-29:
@ZhangGaoxing we're figuring out the process, we know for a fact we do not want System.Device.Gpio to grow indefinitely with all new drivers as users mostly need a single driver but at the same time we do want to support it so we need to figure out better way of support (either through extensions or different nuget). I do recommend you to add a readme page in our repo somewhere that this driver is available from 3rd party github for now and link to your repo.

DanieleNeo, 2020-12-07:
Hi Zhang,

 

sorry for my late reply, I have been very busy this period. It is a pity that there are such problems in supporting OrangePI hardware, no problem to help you testing on Orange Pi PC, time permitting.

 

D.

krwq, 2020-12-10:
@Ellerbach @joperezr @pgrawehr we should prioritize unblocking this

Ellerbach, 2022-12-21:
@DanieleNeo if you want to add this driver, you can do it here:
https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers

As it seems it's based on the Sunxi Driver, check this out: https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers/Sunxi

And just create a new driver called `OrangePiPcDriver`. Test all this, submit a PR and all should be fine! I don't have the hardware to test it but been using a Orange PI Zero2 all so far, things seems to work as expected.

Ellerbach, 2023-01-21:
Closing as done.

closed 2023-01-21 by Ellerbach