dotnet/iot #1229: Who has a demo to control the sg90 steering gear?

bbhxwl, 2020-10-27:
Who has a demo to control the sg90 steering gear?

Ellerbach, 2020-10-27:
> Who has a demo to control the sg90 steering gear?

You have all what you need here: https://github.com/dotnet/iot/tree/master/src/devices/ServoMotor
From the doc:
Position "0" (1.5 ms pulse) is middle, "90" (~2 ms pulse) is all the way to the right, "-90" (~1 ms pulse) is all the way to the left.
So you can find the settings for your servo

bbhxwl, 2020-10-27:
> You have all what you need here: https://github.com/dotnet/iot/tree/master/src/devices/ServoMotor
> From the doc:
> Position "0" (1.5 ms pulse) is middle, "90" (~2 ms pulse) is all the way to the right, [...]
Can let sg90 move, but the angle is always not very accurate. Do not know why?

    var pwm = PwmChannel.Create(0, 0, 50);
            
            ServoMotor servoMotor = new ServoMotor(pwm );
            servoMotor.Start();  
            
         
            servoMotor.WritePulseWidth(1000); // 1ms; Approximately 0 degrees.
            servoMotor.WritePulseWidth(1500); // 1.5ms; Approximately 90 degrees.
            servoMotor.WritePulseWidth(2000); // 2ms; Approximately 180 degrees.

            servoMotor.Stop();

bbhxwl, 2020-10-27:
> > Can let sg90 move, but the angle is always not very accurate. Do not know why?
> 
> This depends of multiple factor, one is if you're using a real hardware PWM which should drive to a good accuracy. Second is the quality of the servo itself.
> In case you're using software PWM, the angle can vary a bit and it's less precise than hardware PWM

I don't know what hardware PWM is. Do you have a picture to explain it?

Ellerbach, 2020-10-27:
> I don't know what hardware PWM is. Do you have a picture to explain it?

sure, check the details here for a RPI: https://github.com/dotnet/iot/blob/master/Documentation/raspi-pwm.md
It's similar if you're using another type of board

bbhxwl, 2020-10-27:
> > I don't know what hardware PWM is. Do you have a picture to explain it?
> 
> sure, check the details here for a RPI: https://github.com/dotnet/iot/blob/master/Documentation/raspi-pwm.md
> It's similar if you're using another type of board

I've enabled it, and NETCORE is much more mature than it used to be with win10 IOT. It was really hard for C to develop raspberry pie. Now there are many examples

Ellerbach, 2020-10-28:
> I've enabled it

So is it working better?

bbhxwl, 2020-10-28:
> > I've enabled it
> 
> So is it working better?

It's normal. Except that the accuracy is not enough, it may be a hardware problem. There is no big problem. thank you

krwq, 2020-10-28:
@bbhxwl I've found the SG90 to not be very accurate, I've once tried having 6x SG90 connected at the same time (using PCA9685) and remember each one of them had slightly different calibration data and it took me a while to get them calibrated reasonably well. In order to calibrate it you need to find pulse width for angle 0 and whichever angle is your max (i.e. 90 degrees), once you have those two values you construct it like following:

```csharp
new ServoMotor(pwmChannel, angleRange, minPulseWidthMicroseconds, maxPulseWidthMicroseconds))
```

I suspect better servos don't have this problems but SG90 seems to be one of the cheapest (I think I got 10 of them for few bucks)

joperezr, 2020-10-28:
Closing issue since it seems the original question was answered. @bbhxwl feel free to ask follow up questions, re-open this issue, or log new ones 😄

closed 2020-10-28 by joperezr