dotnet/iot #1234: Existing designs cannot add GPIO drivers externally

ZhangGaoxing, 2020-10-29:
#1130 Before writing the driver, I refer to the implementation of the existing driver. Due to the protection of some keywords, I cannot call the internal methods of `UnixDriver` outside the `System.Device.Gpio` package.

krwq, 2020-10-29:
Hey @ZhangGaoxing can you elaborate on what should we change in System.Device.Gpio to make it easier for writing drivers like that?

krwq, 2020-12-28:
@ZhangGaoxing is this unblocked now? Can you tell what's blocked still?

ZhangGaoxing, 2020-12-30:
In fact, there is still this problem. So I inherited the `SysFsDriver` class directly. 

Here is a brief description of the error code.
```C#
// namespace Iot.Device.Gpio.Drivers
public unsafe class SunxiDriver : GpioDriver
{
    protected SunxiDriver()
    {
        _sysFSDriver = UnixDriver.Create();
    }

    protected override void ClosePin(int pinNumber)
    {
        // Error: CS1540
        _sysFSDriver.ClosePin(pinNumber);
    }
}
```

pgrawehr, 2020-12-30:
True, creating "forwarding drivers" (drivers that use other drivers) is not possible right now due to protection levels. 

@joperezr This could only be solved by changing all the methods on GpioDriver to public, I suppose. Since that's not what we generally want, I currently see little possibility of fixing this. The use case is valid, though, as we actually do the very same internally already.

krwq, 2021-01-04:
I'm personally in favor of making all `protected internal` public

Ellerbach, 2021-01-04:
> I'm personally in favor of making all `protected internal` public

I agree. This will allow those kind of scenarios. And it will still keep everything super easy and straight forward for anyone who wants to use a simple GpioController without knowing the platform it's running on

pgrawehr, 2021-01-04:
I basically agree as well, but we have to consider the risk that people start using the drivers directly, for which they're not intended. There must be some protection against directly working with the drivers.

joperezr, 2021-01-13:
@pgrawehr has pointed out the issue with doing so, which is people will start using the drivers instead of the controllers, or just get confused with both having similar API surface. It was a very intentional design to make them protected internal. I would much rather move all drivers into System.Device.Gpio instead of making those members public. The controller does a bunch of pin management plus important checks that the driver doesn't do, so using the drivers directly is definitely not intended.

joperezr, 2021-01-13:
To solve @ZhangGaoxing issue, one thing that could be done is that instead of having the pass-through driver call another driver directly, it should instead call a controller that uses that driver underneath, that way it is able to call all required methods.

krwq, 2021-01-14:
@joperezr wouldn't that imply that GpioDriver is an implementation detail in System.Device.Gpio and it has no purpose for anyone implementing things outside of System.Device.Gpio?

joperezr, 2021-01-14:
Yes and no. Having GpioDriver abstract class in S.D.Gpio is nice because then it is an extensibility point that other folks can use if they want to write a complete new driver from scratch outside the library and they can just use that to initialize the controller. The reason why the derived GpioDrivers are public (meaning RaspberryPi3Driver, SysFsDriver, etc) is to allow consumers to manually select which driver to use in case they don't want the one that would be automatically selected for them. In hindsight, I think the right way to do this might have been just having an enum to force a manual selection of a driver, without having to expose all of the derived drivers, but that ship has sailed.

Ellerbach, 2021-02-25:
[Triage] @ZhangGaoxing are the existing work around acceptable?

PR #830 (2019-11-05): Update Hcsr04.cs
Added some comments to make more clear the process of taking a reading

- If PR is fixing any issue please add `Fixes #1234` as a first thing in the description of your PR
- Describe what is the issue being fixed
- If there is any follow-up work please create issues for that
- If your PR is adding device binding please make sure to read [conventions for devices APIs](https://github.com/dotnet/iot/blob/master/Documentation/Devices-conventions.md)
- Avoid force pushing to your PR (especially on large PRs) - it makes it much harder to incrementally review
PR #858 (2019-11-21): Fixed spelling in SoftwarePwmChannel
Fixed a spelling mistake in ArgumentOutOfRangeException message when setting Frequency

- If PR is fixing any issue please add `Fixes #1234` as a first thing in the description of your PR
- Describe what is the issue being fixed
- If there is any follow-up work please create issues for that
- If your PR is adding device binding please make sure to read [conventions for devices APIs](https://github.com/dotnet/iot/blob/master/Documentation/Devices-conventions.md)
- Avoid force pushing to your PR (especially on large PRs) - it makes it much harder to increment [... cut]
PR #1029 (2020-03-31): Update ReadMe Issue Fixes #1022
This addresses the issue wrt 1-wire not working with IoT-Core. Specifically for the RPi and DHT22. This replaces the previous update on the matter by myself.

- If PR is fixing any issue please add `Fixes #1234` as a first thing in the description of your PR
- Describe what is the issue being fixed
- If there is any follow-up work please create issues for that
- If your PR is adding device binding please make sure to read [conventions for devices APIs](https://github.com/dotnet/iot/blob/master/Documentation/Devices-conventions.md)
- Avoid force pushing to your P [... cut]
PR #1371 (2020-12-21): Found some problems in the UFirePh.cs code and made some fixes.
These methods were infinitely recursive:
GetCalibrateHighReference
GetCalibrateHighReading
CalibrateProbeLow
GetCalibrateLowReference
GetCalibrateLowReading

Method PhToMillivolts renamed to PhToEp and now returns ElectricPotential.

Method MVtopH renamed to EpToPh and now takes a parameter of type ElectricPotential.

- If PR is fixing any issue please add `Fixes #1234` as a first thing in the description of your PR
- Describe what is the issue being fixed
- If there is any follow-up work please create issues for that
- If your PR [... cut]
closed 2022-09-30 by msftbot[bot]