dotnet/iot #1239: Support for `System.Device.Dac`

networkfusion, 2020-10-30:
**Is your feature request related to a problem? Please describe.**

nanoFramework supports DAC devices under the `System.Device.Dac` namespace. Given that there is work in progress to align other peripheral interfaces, it would be good to come up with a common implementation.

**Describe the ideal solution**

As implemented in https://github.com/nanoframework/lib-System.Device.Dac or discussion as to how it can be changed to meet the needs of both worlds.

joperezr, 2020-10-30:
FYI: @Ellerbach

Ellerbach, 2020-10-31:
Thanks @networkfusion for starting this discussion. So far we don't have this DAC notion in .NET IoT. And except this closed PR #723 which didn't merge, we had nothing really about Digital to Analog Converter. 
What we have which gets close to this is Audio support https://github.com/dotnet/iot/tree/master/src/devices/Media/SoundDevice using libasound.
That doesn't mean that it's not a scenario that is interesting. And that having a set of design to align and start with is something that clearly help. It's clear that if there is anything coming which will get close to DAC, we will definitely look at the API and try to best align. Get ready to be pulled into the discussion.

krwq, 2020-11-02:
Yes, we do not have DAC abstraction but we do use DAC in our Mcp3008 tests (PWM output with hardware low pass filter made of capacitor)

Ellerbach, 2021-02-25:
[Triage] As soon as we will find a use case and we will have a device to support this, we will clearly work on reusing as much as possible the work done by nanoFramework.

krwq, 2025-07-31:
[Triage] Closing because it had no activity in the last 4 years

closed 2025-07-31 by krwq