dotnet/iot #1241: Mpu9250 - This device does not contain the correct signature 0x48 for a AK8963

FM1973, 2020-11-01:
Hello.

I´m using the mpu9250 in one of my projects. 
I´m using a rasperry pi 3B+ and the latest raspbian os (raspios-buster-armhf-lite 2020-08-20).

When I start and stop my application multiple times (letting it run for a couple of seconds), it sometimes happens, that I get the following exception:

"Unhandled exception. System.IO.IOException: This device does not contain the correct signature 0x48 for a AK8963"

Unfortunately there is no reliable way to reproduce the problem. It just happens sometimes.

```csharp
var i2cSettings = new I2cConnectionSettings(0x01, Mpu9250.DefaultI2cAddress);
_i2CMpu9250 = new Mpu9250(I2cDevice.Create(i2cSettings));
```

**Versions used**

Add following information:

- Version od dev-machine: .net core 3.1.9
- Version on RP3: .net core 3.1.8
- Version of `System.Device.Gpio` package: 1.1.0-prerelease.20276.1
- Version of `Iot.Device.Bindings` package: 1.1.0-prerelease.20276.1

pgrawehr, 2020-11-01:
I've seen similar issues before. Please try to run `i2cdetect -y 1` several times in a row to see whether the MPU9250 stays on the same address all the time. If for some reason, it changes intermittenly to its alternate address (0x49 I think), then that's the reason for the problem. To work around it, make sure that _all_ pins of the chip are connected, even though the documentation may say otherwise. Some of the pins are normally unused in I2C mode, but they serve as address selection or SPI CS pins, and when they float some noise may cause the chip to change its address or switch off completely.

FM1973, 2020-11-01:
Thanks a lot.
I will check that.

Ellerbach, 2020-11-02:
> Unhandled exception. System.IO.IOException: This device does not contain the correct signature 0x48 for a AK8963

You have to access the AK8963 thru the MPU9250. See the example: https://github.com/dotnet/iot/blob/master/src/devices/Mpu9250/samples/Mpu9250.sample.cs

The direct access is not allowed by default and doesn't seems to work in the proper way. The MPU9250 is acting like an intermediate with the AK8963. In theory, when deactivating the MPU9250 feature of being primary device vs AK8963, it should be possible to have a direct access to the AK8963 in the same I2C line. 

Please also check if the fix proposed for #1140 is working. I still don't have the device to do the final check and fix the calibration.

krwq, 2020-11-02:
@Ellerbach if this is in fact the solution for this issue I'd recommend to add some note in the readme of AK8963 to mention MPU9250 case

FM1973, 2020-11-02:
@pgrawehr I´ve run 'i2cdetect -y 1' multiple times after the exception occured. The Mpu9250 is still there with the correct address.

@Ellerbach I´m not even using the magnetometer in my code. The exception is being thrown in the contructor of the MPU9259 class. 
Part of MPU9250 constructor;

``` c#
_ak8963 = new Ak8963(i2cDevice, new Ak8963Attached(), false);
            if (!_ak8963.IsVersionCorrect())
            {
                // Try to reset the device first
                _ak8963.Reset();
                // Wait a bit
                if (!_ak8963.IsVersionCorrect())
                {
                    // Try to reset the I2C Bus
                    WriteRegister(Register.USER_CTRL, (byte)UserControls.I2C_MST_RST);
                    Thread.Sleep(100);
                    // Resetup again
                    WriteRegister(Register.USER_CTRL, (byte)UserControls.I2C_MST_EN);
                    WriteRegister(Register.I2C_MST_CTRL, (byte)I2cBusFrequency.Frequency400kHz);
                    // Poorly documented time to wait after the I2C bus reset
                    // Found out that waiting a little bit is needed. Exact time may be lower
                    Thread.Sleep(100);
                    // Try one more time
                    if (!_ak8963.IsVersionCorrect())
                    {
                        throw new IOException($"This device does not contain the correct signature 0x48 for a AK8963 embedded into the MPU9250");
                    }
                }
            }
```

Ellerbach, 2020-11-02:
> The Mpu9250 is still there with the correct address.

Then it means this setup is different than the one where the AK8963 is embedded into the MPU9250. 
In this case, rather than creating it as an attached device, you can directly create it as a normal device. And that should just work.
Please confirm it's the case, in this case, I'll update the code to add new option in the constructor to create it as a separate device. Good news, it should just work if you create the AK8963 like that. Sorry for the inconvenience, seems like multiple version with different configuration of the MPU9250 are shipped.

Ellerbach, 2020-12-23:
Closing this one as it should be fixed now.

closed 2020-12-23 by Ellerbach