dotnet/iot #1249: Make DelayHelper or similar API part of System.Device.Gpio

hugener, 2020-11-03:
**Timing is often important when developing device bindings, I request that System.Device.Gpio provides something like DelayHelper allowing us to use waits without having to provide our own implementation.**

I am developing my own bindings, some which I don't think will live up the quality standards of Iot.Device.Bindings, but I would still like to base them on System.Device.Gpio and think that it should provide all generic functionality needed to implement a device bindings. An API for delays is missing in System.Device.Gpio and it would be nice to have one that waits the best possible way on the various platform, or just DelayHelper if that's considered good enough.

Ellerbach, 2020-11-04:
> DelayHelper if that's considered good enough

Yes, that's something in the radar, the request came few times and we should have it public with maybe few adjustments. So please stay a bit patient.

hugener, 2020-11-14:
Great looking forward for that!
Is there are place where I can see what else is on the radar?
I found this: https://github.com/dotnet/iot/blob/master/Documentation/roadmap.md, but it wasn't updated since October 2019. Why does it mention 3.0 release when version 1.1 just went out?

I have a ton of questions ranging from testing, HW support, API- and design proposals and don't know if I should create separate question issues or rather gather them in one?

pgrawehr, 2020-11-15:
Yea, the roadmap is very outdated. We're just in the process of getting on track again (after the project has been sleeping a bit due to higher priority issues). Ignore that 3.0 number there, that was a project milestone that should probably never have been named like that. 

It is ok to create issues, as long as it is reasonable that they can be addressed in a timely manner. That will (or rather should) of course be the case for bugs and problems. But if you ask for support for some obscure hardware that nobody has, that will probably not be helpful.

krwq, 2020-11-19:
Yes when we say 3.0 we meant .NET Core 3.0 because we align releases.. Yes the doc is very old and we will publish new one once it's finalized.

@hugener to move this forward sooner I suggest to write specific place and API names you'd like this to have, once we have some agreement in the issue then we can talk about it during our weekly meeting and possibly approve and move forward with publishing these APIs. Perhaps renaming DelayHelper to Delay and some better names and namespaces would suffice here

Ellerbach, 2021-02-25:
[Triage] We will make the DelayHelper public. Work to be done on the naming and namespace to be `Iot.Device`.
While making changes on the helpers, would be perfect to name the `Iot.Device.Common` to just `Iot.Device` for the weather 
helper.
cc @MarkCiliaVincenti

krwq, 2023-01-12:
[Triage] We should make it public but it should stay in Iot.Device.Bindings