dotnet/iot #1250: Question: Why aren't packages released to NuGet.org more often?

hugener, 2020-11-03:


joperezr, 2020-11-03:
Good question, we are going to ship our 1.1 release this month and we have been in talks in order to release stable versions much more often. In the past we were aligning with .NET Core shipping cycles, but there is no reason to continue to do that now so we will move to a much faster cadence (1 new release every 1-2 months).

joperezr, 2020-11-03:
Closing the issue as the question is answered, but feel free to reopen if you have any follow ups.

joperezr, 2020-11-03:
FYI: @richlander @krwq @Ellerbach

closed 2020-11-03 by joperezr

dominikjeske, 2020-11-11:
I can see 1.1 nuget but no release on this page and no release notes?

joperezr, 2020-11-11:
We just shipped it yesterday in NuGet and are working on tagging and finishing up the release notes now 😄

joperezr, 2020-11-11:
Just pushed the release notes: https://github.com/dotnet/iot/releases/tag/1.1

richlander, 2020-11-12:
I'm also hoping that we'll start shipping 1.2 previews pretty soon, hopefully this month.