dotnet/iot #1256: Add documentation guidelines on naming properties and arguments to use when referring to concepts from datasheets

joperezr, 2020-11-05:
This is specifically when referring to concepts from datasheets. This issue will also track the work of updating existing bindings (even if that means breaking changes) to match these new guidelines.

The guideline agreed upon is: In most cases we want to use descriptive names to make the API more reader friendly (specially by people that are not familiar with the datasheet) and just use the triple slash comments to include any relevant datasheet names.

cc: @richlander @krwq @HumJ0218 @Ellerbach 
FYI: @pgrawehr @shaggygi @ZhangGaoxing

shaggygi, 2020-11-05:
"specially by people that are not familiar with the datasheet"

I see this from both sides, but in most cases, devs that will be using these APIs will most likely have to reference datasheets at some point (even to only wiring the pins).  I agree with @HumJ0218 and believe bindings should using datasheet convention naming (pin, registers, etc.) especially for more advanced bindings.  You could have both where you have he lower level calls to configure stuff and then higher level methods that offer better descriptions that call the lower levels.

I guess examples would help regarding guidance on how and what to write in different scenarios.  For example, if you have a simple binding like ReadTemperature() that is on a binding with just PWR/GND/Signal pins... then description pattern is better/simple.  However, if the binding has different registers required to be configured to perform the main end call, then datasheet pattern is better.  Having the ability to pass in the bits/bytes to specific registers offer more flexibility to advanced users.  In addition, bindings that have registers with the different bit flags and such most likely offer multiple functionality and would be difficult to offer in a smaller set of methods (unless overloaded).

Lastly, searching is another reason I like the approach of using the datasheet naming instead of sometime guessing the descriptive name.  It makes it easier to find the info referenced throughout the entire datasheet and sometimes the internet for finding solutions on StackOverflow.  Use the Mcp23xxx binding/datasheet for example.  What would be the naming guidance for register GPPU?  PullUpResistorConfiguration?  These names could get very long and we would be leaving it up for binding creators to interpret the best name.

```c#
// very simplified example...

public void WriteRegister(byte gppu) { ... }

// or 

public void WriteRegister(byte pullUpResistorConfiguration) { ... }
```

richlander, 2020-11-05:
Here are some good examples to discuss:

- https://github.com/dotnet/iot/blob/master/src/devices/ShiftRegister/ShiftRegisterPinMapping.cs#L23
- https://github.com/dotnet/iot/blob/master/src/devices/Sn74hc595/Sn74hc595PinMapping.cs#L20
- https://github.com/dotnet/iot/blob/master/src/devices/Mbi5027/Mbi5027PinMapping.cs#L20

The first one is an concrete but least-common-denominator binding for shift registers (no specific data sheet). The second two are specific. I went with good names for the first one, and data sheet names for the other two. I had a strong opinion about that at the time. I'm no longer sure.

shaggygi, 2020-11-06:
"Here are some good examples to discuss:"

Yes, I agree with using description naming seems to be a good example in this case.

HumJ0218, 2020-11-06:
Although I prefer to use datasheet pin names as parameter names, there are exceptions.

For example, the three input pins of the 74595 are confusing me. In some documents they are `SRCLK`, `RCLK`, `SER`, sometimes `SH_CP`, `ST_CP`, `DS`, and other descriptions. If you name it according to the description parameters described in the documentation, it is difficult to determine which name to use unless you use vendor-specific models such as TI_SN74HC595, PHILIPS_74HC595. The functions of these pins are the same, and it may be easier to use specific functions as parameter names.

fehdem, 2020-11-07:
This is a very interesting discussion. On one hand I appreciate APIs with names requiring no further explanation w. r. t. there function and the detail behind. In context of a binding this makes binding and device kind of black box I don't need to care about. When I invoke something called GetTemperature I know what I get, I can focus on my use case and stop further thinking.
On the other hand we're dealing here with devices which may require a larger extend of configuration and controlling. If the device addresses a broad variety of applications and use cases it is less clear at which point of abstraction the API should be, I think. 
If the level of abstraction is too high, I could presume to much of an application and limit the generality of the binding. If I the abstraction is too low I could stay with a register definition and basic I/O functions. Neither will satisfy the need of our binding users.

I just started with a new binding for a programmable PLL clock generator. At this point I'm thinking about the API design, but I didn't make my mind, yet. 
Ideally I would offer two levels of abstraction. One rather highly level, with basic functions, for instant / hobbyist use. And a lower abstract for those who really need to get to the details of the chip and its variants. 
Maybe it would be even helpful to have dedicated classes for the two abstraction levels. 

Conclusion: I think the sweet spot of abstraction depends on the actual device and its use cases. A two level abstraction approach as a general design approach could be one solution.

richlander, 2020-11-07:
@fehdem -- This conversation isn't really about method names since the datasheets typically don't define those. It's more about parameter names and pin mappings, and to a lesser degree property and field names.

fehdem, 2020-11-07:
@richlander Ah, OK. I got this wrong.

Ellerbach, 2020-11-08:
I'm a big fan of using as descriptive names as possible. And I think @HumJ0218 write it well:

> For example, the three input pins of the 74595 chips are confusing me. In some documents they are `SRCLK`, `RCLK`, `SER`, sometimes `SH_CP`, `ST_CP`, `DS`, and other descriptions

And from my experience, it's always much easier to find back in the datasheet what is the abbreviation from a long descriptive name than the opposite. Reason as well is that like the case of 74595 described, once you get multiple long names, you can figure out much easier multiple short ones and that makes the life easier.

And we had already in [this section](https://github.com/dotnet/iot/tree/master/src/devices#naming-of-functions) the recommendation to avoid abbreviations and use long names.