dotnet/iot #1261: .vsts-ci.yml looks stale

richlander, 2020-11-06:
Some of the versions seem stale.

* `image: microsoft/dotnet-buildtools-prereqs:ubuntu-14.04-cross-0cd4667-20170319080304`
* `name: Hosted VS2017`
* `Hosted Ubuntu 1604`

Those values should all be something newer

I also see `release/3.0`. That branch looks dead at this point.

rich

joperezr, 2020-11-06:
the branch we probably want to keep just in case we ever submit a PR there, in fact I'll likely replace it with release/*
as for the machine pools I'll check what is it that we are using in dotnet/runtime these days and make sure we are using the same ones but usually I don't touch these unless I get pinged by the infrastructure team saying they plan on deprecating some pools

richlander, 2020-11-06:
Related ... this value doesn't look correct: https://github.com/dotnet/iot/blob/00f9da195e6c3bb296450ab88303e031ad55aff8/eng/sendToHelix.proj#L15

I'm trying to upgrade the tests to 5.0 but am running into an invisible wall that I cannot figure out as part of https://github.com/dotnet/iot/pull/1260. I tried to change that value but it doesn't seem to be helping.

joperezr, 2020-11-09:
Yeah I don't think Helix is ready for us to move to 5.0 just yet. We use the xunit runner which today cross-compiles for netcoreapp 2.0 which is why we are setting that property you found there to run the tests. I'll ping the infrastructure team to check what is the Helix SDK support for net5.0

closed 2021-04-15 by joperezr