dotnet/iot #1265: Documentation for CharacterLcd not up to date

Ellerbach, 2020-11-08:
Documentation of main Readmed for CharacterLcd is not up to date.
Refer to #1264 
The bindings has been adjusted and LcdRgb1602 is now replaced by LcdRgb which takes new arguments.

This is not correct:

```csharp
var i2cLcdDevice = I2cDevice.Create(new I2cConnectionSettings(busId: 1, deviceAddress: 0x3E));
var i2cRgbDevice = I2cDevice.Create(new I2cConnectionSettings(busId: 1, deviceAddress: 0x62));
using (var lcd = new LcdRgb1602(i2cLcdDevice, i2cRgbDevice))
{
    lcd.Write("Hello World!");
    lcd.SetBacklightColor(Color.Azure);
}
```

Correct should be:

```csharp
var i2cLcdDevice = I2cDevice.Create(new I2cConnectionSettings(busId: 1, deviceAddress: 0x3E));
var i2cRgbDevice = I2cDevice.Create(new I2cConnectionSettings(busId: 1, deviceAddress: 0x62));
using (var lcd = LcdRgb(new System.Drawing.Size(16, 2), i2cLcdDevice, i2cRgbDevice))
{
    lcd.Write("Hello World!");
    lcd.SetBacklightColor(Color.Azure);
}
```

pgrawehr, 2020-11-09:
Ups. Probably my fault. Sorry.

Ellerbach, 2020-11-09:
> Ups. Probably my fault. Sorry.

Don't be sorry :-) It's all ok! Just go to the proposed PR and check if all is ok :-)

PR #1266 (2020-11-12): Fixing documentation for Grove RGB
- Fixes #1265 
- Related to #1264
closed 2020-11-12 by Ellerbach