dotnet/iot #1267: Raspberry Pi 4 Model 0b3112 & 0c112 Not supported in System.Device.Gpio

Richy1989, 2020-11-08:
**Descritpion**
Raspberry Pi 4 Model "b03112 (Sony, UK)" Missing. See: https://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/
--

**How to reproduce**
Only works with Raspberry Pi Model "3112":

Force Driver during controller initialization. 
```csharp
var gpioController = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
```
Will throw an exception that no supported Raspberry Pi module is found. 
**Expected behavior**

Should initialize the GPIO controller. 

**Actual behavior**

throws an exception that no supported Raspberry Pi plattform is found. 

**Proposed Fix**
Add model to : iot/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs
```csharp
                case 0x3111:
                case 0x3112: // <--- add this line to support Raspberry Pi 4 Model b03112 and c03112. 
                    return Model.RaspberryPi4;
```

**Versions used**

- `dotnet --info` on the machine being used to build
.NET SDK (reflecting any global.json):
 Version:   5.0.100-rc.2.20479.15
 Commit:    da7dfa8840

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.19041
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.100-rc.2.20479.15\

Host (useful for support):
  Version: 5.0.0-rc.2.20475.5
  Commit:  c5a3f49c88

.NET SDKs installed:
  5.0.100-rc.2.20479.15 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.0-rc.2.20475.17 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.21 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.0-rc.2.20475.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.7 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.0-rc.2.20475.6 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
.NET SDK (reflecting any global.json):
 Version:   5.0.100-rc.2.20479.15
 Commit:    da7dfa8840

Runtime Environment:
 OS Name:     ubuntu
 OS Version:  20.04
 OS Platform: Linux
 RID:         ubuntu.20.04-arm64
 Base Path:   /usr/share/dotnet/sdk/5.0.100-rc.2.20479.15/

Host (useful for support):
  Version: 5.0.0-rc.2.20475.5
  Commit:  c5a3f49c88

.NET SDKs installed:
  5.0.100-rc.2.20479.15 [/usr/share/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.0-rc.2.20475.17 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.0-rc.2.20475.5 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
- Version of `System.Device.Gpio` package
1.1.0-prerelease.20555.1
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

krwq, 2020-11-09:
@Richy1989 good find, do you want to send a PR with a fix?

Richy1989, 2020-11-09:
Hey, 
I created a PR. See: https://github.com/dotnet/iot/pull/1269

PR #1269 (2020-11-10): #1267 added support for Raspberry Pi 4 Model 0b3112 & 0c3112 in System.Device.Gpio
Fixes #1267

Added support for Raspberry Pi 4 Model 0b3112 & 0c3112 in System.Device.Gpio
closed 2020-11-10 by krwq