dotnet/iot #1270: GpioController is not correctly identifying Raspberry Pi with Revision c03112

Semour81, 2020-11-09:
I would like to use Dht11 module on Raspberry PI 4 Revision 1.2 (Model C03112). 

**Steps to reproduce**

```csharp
using (Dht11 dht = new Dht11(17))
{
    var temperature = dht.Temperature;
}
```

**Expected behavior**

Accessing dht.Temperature must work without getting an exception. Used PI model supports InputPulUp pin mode.

**Actual behavior**

Exception occurs when accessing dht.Temperature.

```
An unhandled exception of type 'System.InvalidOperationException' occurred in System.Device.Gpio.dll: 'Pin 17 does not support mode InputPullUp.'
```

**Further analysis**

In [Issue 1145](https://github.com/dotnet/iot/issues/1145) there is a workaround described:

```csharp
GpioDriver driver = new RaspberryPi3Driver();
GpioController controller = new GpioController(PinNumberingScheme.Logical, driver);
```

Unfortunately this also does not help because PI model could not be detected and RaspberryPi3Driver() cannot be initialized.

```
An unhandled exception of type 'System.PlatformNotSupportedException' occurred in System.Device.Gpio.dll: 'Not a supported Raspberry Pi type: Unknown'
at System.Device.Gpio.Drivers.RaspberryPi3Driver..ctor() at MyConsoleApp.Program.ReadFromDHT11()
```
The actual reason of that behavior is, that enum Model in System.Device.Gpio.RaspberryBoardInfo does not contain C03112 value.

Model C03112 is officially listed on [Raspberry.org](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), model C03114 seems also not supported.

**Versions used**

_Raspberry:_

/proc/cpuinfo:
Hardware: BCM2711
Revision: c03112
Serial : 10000000683cf6ea
Model : Raspberry Pi 4 Model B Rev 1.2

_OS:_
Linux 5.4.0-1022-raspi #25-Ubuntu SMP PREEMPT Thu Oct 15 14:22:53 UTC 2020

_Dotnet:_

.NET Core SDK (reflecting any global.json):
 Version:   3.1.403
 Commit:    9e895200cd

Runtime Environment:
 OS Name:     ubuntu
 OS Version:  20.04
 OS Platform: Linux
 RID:         ubuntu.20.04-arm
 Base Path:   /usr/bin/dotnet/sdk/3.1.403/

Host (useful for support):
  Version: 3.1.9
  Commit:  774fc3d6a9

.NET Core SDKs installed:
  3.1.403 [/usr/bin/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.9 [/usr/bin/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.9 [/usr/bin/dotnet/shared/Microsoft.NETCore.App]

_Libraries:_

Iot.Device.Bindings: 1.1.0-prerelease.20555.1
System.Device.Gpio: Version="1.1.0-prerelease.20555.1

Ellerbach, 2020-11-09:
I see. Seems that this line is causing the issue:https://github.com/dotnet/iot/blob/de328cacc0dd03d539d6508c6d6239383c44af1b/src/devices/Dhtxx/DhtBase.cs#L165.
I've seen this issue on few other boards as well which doesn't support input pull up.
Best is for sure to change this to a normal Input (supported everywhere), the pull up has to be external (which is the case 99.9999% of the time with those sensors). 
So if you pull the code from the repo and adjust this line, it should work. The time we make a fix and push it. Would be perfect if you can tell us if that fixed the issue.

sensslen, 2020-11-10:
I'm not sure, but if I read correctly the actual request is to support more revision numbers of the Raspberry pi.

Semour81, 2020-11-10:
To solve this issue adding C03112 to revision numbers in System.Device.Gpio.RaspberryBoardInfo is enough. However, I would also suggest to add C03114 since this is also missing. 

Not using PinMode.InputPullUp is another thing, in this case it won't solve anything since Rasbperry BI model is not recognized at all.

Ellerbach, 2020-11-14:
Related to #1291

krwq, 2020-12-18:
@pgrawehr @Ellerbach do you know about any other values we're missing? I've seen this https://github.com/dotnet/iot/issues/1291 but not sure there are any specifics mentioned

pgrawehr, 2020-12-18:
@krwq: There's a link in the OP above. The page on RaspberryPi.org seems to be current, as it includes both the Rasppi 4B 8GB and the Raspi 400.

closed 2020-12-21 by krwq