dotnet/iot #1272: Dispose not handled correctly

richlander, 2020-11-09:
Example: https://github.com/dotnet/iot/blob/master/src/devices/Ak8963/Ak8963.cs#L334-L341

The `null` assignment should be outside the `if` check. The null assignment removes a ref count on the object, for the GC.

pgrawehr, 2020-11-09:
I believe there are also many cases where the null asignment is missing completely (not to say that there are no ObjectDisposedExceptions either).

Related question: Should readonly declarations be preferred or a null assignment in dispose?

krwq, 2020-11-10:
Also we should unify names like autoDispose -> shouldDispose as part of this. Perhaps we should consider writing an analyzer for this so it's enforced?

Ellerbach, 2020-11-12:
> autoDispose -> shouldDispose

I think there are only few left as far I can remember. That's the reason I wrote the Binding do/doesn't with concrete example and few patterns. Now with the nullability and few other improvement, the list should be updated!

richlander, 2020-11-13:
I think I got the last of the autoDispose in one of my last PRs.

> Related question: Should readonly declarations be preferred or a null assignment in dispose?

I think a null assignment. It is better for garbage collection to null the reference. Also, readonly doesn't seem to be buying us much in this case now that we have NRTs to help us.

krwq, 2022-11-17:
Possible solution: https://github.com/dotnet/iot/issues/1971