dotnet/iot #1291: Release 1.1 does not support InputPullUp/InputPullDown mode for RaspberryPi2B

martinsymon, 2020-11-13:
I don't know if this is really bug but with release 1.1 I'am no longer able to set pin driver mode to InputPullUp value on my Pi Model 2B v1.1 rev. a01041. 

**Steps to reproduce**

```csharp
GpioController controller = new GpioController();
controller.OpenPin(17, PinMode.InputPullUp);
```

**Expected behavior**

When I run this code in release 1.0 everythink works fine.

**Actual behavior**

When I run this code in release 1.1 I get exception "Pin 17 does not support mode InputPullUp".

I debugged the source code of 1.1 and my RPi2 gets LibGpioDriver. When setting DriveMode.InputPullUp there is condition for gpiod v1.5. But RPi2 has up-to-date version only 1.2 so I get mentioned exception. Will be added RPi2 PullUp/PullDown support in future or do I have to switch to newer RPi? Or should I stick with release 1.0 for my antique board?

Thanks guys. Keep up good job.

pgrawehr, 2020-11-13:
Do you know (or can you find out) which GpioDriver your board was using previously? 

There is (and never was) a driver for Raspberry Pi 2's, so it probably used the SysFs driver. This also does not support the pull ups, but probably didn't bother about it.

joperezr, 2020-11-13:
THanks for logging the issue @martinsymon! Yeah I'm also curious about that, not sure how it was working before, so if you could it would be interesting to see which driver was being used before as the only driver that supported pull-up and pull-down on 1.0 was the RaspberryPi3 one, which I doubt was the one getting picked. Also, can you try installing libgpiod-dev package in your Pi? (apt-get install libgpiod-dev) I ask because seems like the latest version of libgpiod is 2.1, not 1.2, but that might just be because your Pi doesn't support 2.1, so I'd be interested to see if that unblocks you. Let us know if there is anything we can do to help!.

ayushagarwal181, 2020-11-13:
I am facing a similar issue in Raspberry Pi 4 Model B

I get the below error -
Unhandled exception. System.InvalidOperationException: Pin 4 does not support mode InputPullUp.
   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)

I even tried to do it by opening the pin controller.OpenPin(pin, PinMode.Input), but no luck.
full code - https://stackoverflow.com/questions/64807557/net-core-iot-no-data-coming-from-seed-studio-grove-dth11-sensor-when-using-wit

joperezr, 2020-11-13:
Interesting, I suspect that this has to do with the logic we have which automatically detects which driver to use since we changed that to 1.1, so that may be causing to pick libgpiod over raspberryPi one which shouldn't happen. Can you both try to initialize the controller like this instead:
```c#
GpioController controller = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
controller.OpenPin(17, PinMode.InputPullUp);
```

joperezr, 2020-11-13:
If that works, then this means there is a bug in that board detection logic which we have to fix.

ayushagarwal181, 2020-11-14:
I tried using the below code

using (GpioController controller = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver()))
{
    controller.OpenPin(4, PinMode.InputPullUp);
}

but i am getting an error during runtime

Unhandled exception. System.PlatformNotSupportedException: Not a supported Raspberry Pi type: Unknown
   at System.Device.Gpio.Drivers.RaspberryPi3Driver..ctor()

I am using Raspberry Pi 4B and i guess the RaspberryPi3Driver is not compatible with it.

ayushagarwal181, 2020-11-14:
I tried using the below code

using (GpioController controller = new GpioController(PinNumberingScheme.Logical, new SysFsDriver()))
{
    controller.OpenPin(17, PinMode.InputPullUp);
}

but i am getting an error during runtime

Unhandled exception. System.InvalidOperationException: Pin 17 does not support mode InputPullUp.
   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)

I installed libgpiod-dev package in my pi and tried with both SysFsDriver & RaspberryPi3Driver, but no luck

pgrawehr, 2020-11-14:
@ayushagarwal181 : The second observation is expected, the sysFsDriver really does not support pullups. However, the Raspberry Pi4 should be supported by the Pi3 driver, I've always used that one. Can you tell us what exact model of the Pi4 you are using and which OS?

ayushagarwal181, 2020-11-14:
@pgrawehr : I am using Raspberry Pi 4 Model B with 2GB Ram and the OS installed is Raspbian GNU/Linux 10 (buster), Architecture arm

pgrawehr, 2020-11-14:
@ayushagarwal181  Looks like something in the board detection is seriously broken. I need to debug this. Can you please post the output of `cat /proc/cpuinfo`? Maybe the board identifier is different from what the specification says (we do not have all possible models of all boards to test).

Ellerbach, 2020-11-14:
Just tested on my RPI4 model B, all worked perfectly as expected. Both the "normal" detection is working, passing the Rpi3 driver is working as well.

@pgrawehr here are the info for my model in case:

```shell
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000003f1299b4
```

Ellerbach, 2020-11-14:
On my RPI4 model B rev 1.2, same issue with `GpioController controller = new GpioController();`
Got exception:

```text
Unhandled exception. System.InvalidOperationException: Pin 17 does not support mode InputPullUp.
   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)
   at helloworld.Program.Main(String[] args) in C:\tmp\TestGrove\TestGrove\Program.cs:line 21
```

with GpioController controller = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());``
Got different exception:

```text
Unhandled exception. System.PlatformNotSupportedException: Not a supported Raspberry Pi type: Unknown
   at System.Device.Gpio.Drivers.RaspberryPi3Driver..ctor()
   at helloworld.Program.Main(String[] args) in C:\tmp\TestGrove\TestGrove\Program.cs:line 20
```

Now the info:

```text
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03112
Serial          : 1000000041ead9f5
Model           : Raspberry Pi 4 Model B Rev 1.2
```

Ellerbach, 2020-11-14:
And this bug has already been addresses but right after the 1.1 realease.
See here: https://github.com/dotnet/iot/blob/c8130309c99a35ce68a10c40ca9003b0261cc2ef/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L182
And here: https://github.com/dotnet/iot/blob/17d497e71147431e12784fcd15e6aaea37a409a3/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L181

Ellerbach, 2020-11-14:
and we may miss few other revisions: https://github.com/dotnet/iot/issues/1270#issuecomment-724514584

martinsymon, 2020-11-14:
> THanks for logging the issue @martinsymon! Yeah I'm also curious about that, not sure how it was working before, so if you could it would be interesting to see which driver was being used before as [...]

Hi, thanks for reply. When I use release 1.0 default picked driver is System.Device.Gpio.Drivers.**RaspberryPi3Driver**. Model of my board is Pi Model 2B V1.1. When I use release 1.1 and try to create GpioController with explicit RaspberryPi3Driver I get exception " System.PlatformNotSupportedException: Not a supported Raspberry Pi type: RaspberryPi2B". Is there some main reason for bannig this board from using of RaspberryPi3Driver?

martinsymon, 2020-11-14:
> Do you know (or can you find out) which GpioDriver your board was using previously?
> 
> There is (and never was) a driver for Raspberry Pi 2's, so it probably used the SysFs driver. This also does not support the pull ups, but probably didn't bother about it.

Thanks for reply. In release 1.0 default assigned driver is System.Device.Gpio.Drivers.RaspberryPi3Driver. Model of my board is Pi Model 2B V1.1.

pgrawehr, 2020-11-15:
@martinsymon : This is clearly a bug. We have changed the way the different types of boards are detected, and unfortunatelly, the test was missing some possible hardware revisions. 

In your case, it even looks like we were under the impression that the Raspberry Pi 2 wouldn't work at all with our driver, so it was omitted. It would be good if you could try to build the repo yourself and add your revision to the board detection code (see @Ellerbach 's comment above). If that makes it work fine, we would now that Pi2B actually works as well.

Ellerbach, 2020-12-23:
@pgrawehr and @krwq I feel this issue could be closed with the latest improvement in the logic detection and release 1.3. Shall we close it?

krwq, 2020-12-23:
@martinsymon can you try 1.3 and see if you still see any issues?

Ellerbach, 2021-03-25:
[Triage] @martinsymon we will close this issue as it seems fixed with one of the latest release. Please feel free to reopen if you still have the issue with the latest 1.4 version.

closed 2021-03-25 by Ellerbach