dotnet/iot #1292: SPI: Unhandled exception. System.Exception: Can't find a PN532

Crashdummyy, 2020-11-13:
**Describe the bug**
Hello Team,

thanks a lot for your really great work here.
I However have one problem I've no idea how to solve and am thus asking for help.

I Can't instantiate PN532 with the message of the title.
The module I use is the [Waveshare PN532 Hat](https://www.waveshare.com/wiki/PN532_NFC_HAT#Raspberry_Pi_examples) 
The [DemoCodes](https://www.waveshare.com/w/upload/6/67/Pn532-nfc-hat-code.7z) appear to work:

<sampleOutput>

```bash
Found PN532 with firmware version: 1.6
Waiting for RFID/NFC card...
................Found card with UID: ['0x4', '0x84', '0x81', '0x6a', '0xda', '0x48', '0x80']
................Found card with UID: ['0x4', '0x83', '0x86', '0x6a', '0x8d', '0x5b', '0x80']
................Found card with UID: ['0x4', '0x33', '0x8b', '0x2', '0xa1', '0x5d', '0x80']
```

</sampleOutput>

**Steps to reproduce**

```csharp
using System;
using System.Device.Spi;
using Iot.Device.Pn532;
using Iot.Device.Pn532.ListPassive;

namespace PiNfc
{
    public class Program
    {
        public static void Main(string[] args)
        {
            var device = new Pn532(SpiDevice.Create(new SpiConnectionSettings(0)));

            if (!(device.FirmwareVersion?.IsPn532 ?? false))
            {
                Console.WriteLine("No PN532 detected... return");
                return;
            }
            
            Console.WriteLine($"Firmware: {device.FirmwareVersion}");
            Console.WriteLine($"Version: {device.FirmwareVersion.Version}");
            Console.WriteLine($"Supported cards: {device.FirmwareVersion.VersionSupported}");
        }
    }
}
```

**Expected behavior**

I at least had hoped to read the desfire chip i put on top of the reader.

**Actual behavior**

System.Exception *Can't find a PN532* thrown inside the Constructor of PN532

**Versions used**

csproj:
```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net5.0</TargetFramework>
  </PropertyGroup>

  <ItemGroup>   
    <PackageReference Include="Iot.Device.Bindings" Version="1.1.0" />   
    <PackageReference Include="System.Device.Gpio" Version="1.1.0" />
  </ItemGroup>

</Project>
```

Dotnet info of machine that is building and runnning the application:
```bash
pi@doorPi:~/tools/PiNfc $ dotnet --info
.NET SDK (reflecting any global.json):
 Version:   5.0.100-rc.2.20479.15
 Commit:    da7dfa8840

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /usr/share/dotnet/sdk/5.0.100-rc.2.20479.15/

Host (useful for support):
  Version: 5.0.0-rc.2.20475.5
  Commit:  c5a3f49c88

.NET SDKs installed:
  5.0.100-rc.2.20479.15 [/usr/share/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.0-rc.2.20475.17 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.0-rc.2.20475.5 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```

pgrawehr, 2020-11-14:
This is likely a problem with the usage of the SPI bus. dotnet/iot by default relies on hardware CS support. if I see correctly the python example application controls the CS line directly from the application. Check https://github.com/dotnet/iot/blob/master/Documentation/raspi-spi.md for information on enabling hardware CS support. For a quick test, you can just hardwire the CS line of the PN532 to ground and see wheter that makes it work.

Crashdummyy, 2020-11-14:
Okay to get this right.
From the documentation of waveshare:
`SPI：We use D4 (BCM) pin as Chip select pin of SPI interface. Take care about GPIO conflict.`

My /boot/config.txt looks like this:
```cfg
#dtparam=spi=on
dtoverlay=spi0-1cs,cs0_pin=4
```

I however still cant get it to work

pgrawehr, 2020-11-14:
Do you have the possibility to attach a scope to your CS pin to see whether it is being used correctly? I've never used it this way, but @Ellerbach may have more ideas about what's wrong.

Crashdummyy, 2020-11-14:
An oscilloscope?
Sorry only an multimeter.

This is my first time working with SPI at all so I might not be able to see the problem.

From the waveshare HP:
[image]

If I see this correctly the NSS -> D4 switch maps the CE0 ( that is the CS, right? ) to GPIO4.
As you suggested first I wired then GPIO4 to Ground where I still didn't get any output.

I even set a wire from GPIO4 to GPIO8 ( CE0 N ) and inside the python example I changed cs=4 -> cs=8
The example still appears to be runnning

pgrawehr, 2020-11-14:
Unfortunately, I don't have such a board, so I can't tell you anything about the specifics of it. But I know that the most common problems with SPI are incorrect CS settings. Yes, CE0 is the CS pin (just a different name for it). 

Can you maybe try to use I2C instead? This is likely to cause fewer configuration problems.

Crashdummyy, 2020-11-14:
Okay I will try that once I get home.
With which Board did you do your tests?

The thing is that I need to put my application to an already wired pi that has an adafruit pn532 wired to spi

pgrawehr, 2020-11-14:
I have not been doing many tests with SPI (and none with a PN532). @Ellerbach has written the binding for this board, so he might be able to help.

Ellerbach, 2020-11-14:
> @Ellerbach has written the binding for this board, so he might be able to help.

I'll have a look as soon as possible. I'll try tomorrow, if not, then the next weekend. I2C and Serial Port work for sure. SPI was experimental: https://github.com/dotnet/iot/tree/master/src/devices/Pn532#current-implementation. So need to see if we have now what's needed to make it fully working. It worked on my device but was not the best experience. The device has a very specific way of working with SPI. Sorry for the inconvenience.

PR #1306 (2020-11-24): Fixing PN532 to work with SPI 
- Fixes #1292
- Fixes bug into TryDecode106kbpsTypeA function introduced during nullability migration
- Improving sample with example for each interface
- Adjusting documentation
closed 2020-11-24 by Ellerbach