dotnet/iot #1322: Automate update work

krwq, 2020-11-25:
We should write a simple tool/script which automates as many steps of creating new version of dotnet/iot:
- bump versions (possibly tool should have: `--update-minor`, `--update-major` or just `--version`
- push package to nuget (credentials would be an input for that tool), alternatively we should create a doc describing this process
- update dependencies after version is up - we do have a bot for that but instead of sending a single PR it sends multiple PRs, perhaps there is a way to tweak it

cc: @joperezr

joperezr, 2020-11-30:
We could add something to the build scripts that automate 1 and 3, for 2 nuget already has a single command line that can push packages to nuget.org that you need to pass credentials too. Given only a few have permissions to do so, I think it is better to follow the UI publish as it lets you validate and preview things before pushing (which is a non-reversible action)

krwq, 2020-12-01:
@joperezr let's start with automating 1 and 3, for number 2 I think it is still worth to push it automatically, the reason is that we would like to eventually push more frequently and manual work will be a bit of a hassle. We can test it on myget first

pgrawehr, 2020-12-18:
I think for 3, dependabot will create only one PR when there's a single place where the version of a specific package is mentioned. We should move all package versions that we use to `/eng/Versions.props`.

Ellerbach, 2021-02-18:
[Triage] Thanks @krwq for volunteering in turning this as documentation :-)

krwq, 2021-02-19:
This thread will be replaced by https://github.com/dotnet/iot/pull/1443

closed 2021-03-25 by Ellerbach