dotnet/iot #1323: Confusing TFM in nuget package

colombod, 2020-11-25:
The `System.Device.Gpio` package targets
- .NET 5.0 
- netstandard2.0

While the `Iot.Device.Bindings` package targets
- .NET 5.0 
- netcoreapp2.1

Can this be aligned to the `System.Device.Gpio` target?

joperezr, 2020-11-30:
This is by design, as it is not a requirement to have them both target the same frameworks. Gpio library is able to have the implementation with only netstandrd 2.0 surface area, while there are many bindings where that is not enough and some types from netcoreapp2.1 are required. We could add an extra ns2.0 implementation on the bindings package that would simply drop support for the bindings that don't work for ns2.0, so those would only be present in nca2.1+

colombod, 2020-12-01:
I haven't checked which bindings classes have the netcoreapp requirements. I would like to suggest splitting the package to have a package for device binding. What do you think?

joperezr, 2020-12-01:
We have considered doing this in the past but haven't done it for several reasons, among which are: simplify our shipping process, the fact that a lot of bindings depend on each other, simpler consumption side, most projects will require more than one binding, better discoverability, among others. We can revisit this to decide if we want to change our decision now but I would initially be more inclined to just add a NS2.0 target to the bindings p ackage with the bindings that are supported in ns2.0

Ellerbach, 2020-12-23:
@colombod to add couple of elements: if you really need to have something portable it's usually because you want to run on mono or other non .NET Core/.NET 5.0 supported platforms. In this case, yes, that would make life easier.
And I've been running binding code on mono on Raspberry Pi 0 by porting the code binding itself to .NET 4.7+ (would have been almost same for nestandard 2.0. It takes a little bit of time, that's true but you just have to do it for the bindings you need.
Readjusting all the repository with all the bindings we have today would be quite some work and would introduce bugs for sure. So I would let the situation as it is. Again, there are very limited scenarios where .NET Core/.NET 5.0 would not work.
PS: feel free to close the issue if you think both @joperezr and me answer are acceptable.

colombod, 2020-12-23:
Makes sense and understand the work implications. I still think this could bring a bigger dependency graph but it is fine to close this issue. Thank you all.

closed 2020-12-23 by colombod