dotnet/iot #1328: Ws2812b LED driver does not work properly on Raspberry Pi 2B and 4B when LED length < 8.

SongweiFu-ITEM, 2020-11-27:
When LED length is smaller than 8, the color of all LEDs are not correct. But when length is larger than 8, all LEDs are again all ok. 

The IoT.Device.Bindings in my NuGet has v1.1.0 (published on 11/9/2020). 
The sample program I took is from here: https://github.com/dotnet/iot/tree/master/src/devices/Ws28xx

**dotnet --info** from my Raspi 2B
.NET Core SDK (reflecting any global.json):
 Version:   3.1.102
 Commit:    573d158fea

Runtime Environment:
 OS Name:     raspbian
 OS Version:  8
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet-arm32/sdk/3.1.102/

Host (useful for support):
  Version: 3.1.2
  Commit:  916b5cba26

.NET Core SDKs installed:
  3.1.102 [/home/pi/dotnet-arm32/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.2 [/home/pi/dotnet-arm32/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.2 [/home/pi/dotnet-arm32/shared/Microsoft.NETCore.App]

Ellerbach, 2021-03-25:
[Triage] Up for grabs!

jesseryoung, 2021-11-12:
Just ran into this today as well. Funnily enough, if you increase the number of pixels to 8 (without actually having 8 pixels on the line) it works just fine.