dotnet/iot #1329: Raspberry Pi Model 3A+ Throws exception when trying to open pin as InputPullUp

Jumble0470, 2020-11-28:
using the System.Device.Gpio assembly fails on Raspberry Pi Model 3A+, same code works as expected on Raspberry Pi Model 4B. 

When trying to open a pin from the GpioController the following exception is thrown:
Unhandled exception. System.PlatformNotSupportedException: This driver is generic so it can not perform conversions between pin numbering schemes.

**Steps to reproduce**

```csharp
// put sample code here
using System;
using System.Device.Gpio;

namespace bugereport
{
    class Program
    {
        static void Main(string[] args)
        {
            var ctl = new GpioController(PinNumberingScheme.Board);
            ctl.OpenPin(15, PinMode.InputPullUp);
            var value = ctl.Read(15);

            Console.WriteLine(value);
        }
    }
}

```

**Expected behaviour**

I would expect the pin value to read and printed .

**Actual behaviour**

The following exception is thrown
Unhandled exception. System.PlatformNotSupportedException: This driver is generic so it can not perform conversions between pin numbering schemes.
   at System.Device.Gpio.Drivers.SysFsDriver.ConvertPinNumberToLogicalNumberingScheme(Int32 pinNumber)
   at System.Device.Gpio.GpioController.GetLogicalPinNumber(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPinCore(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)
   at bugereport.Program.Main(String[] args) in /home/pi/bugereport/Program.cs:line 11

**Versions used**

 <PackageReference Include="Iot.Device.Bindings" Version="1.2.0" />
 <PackageReference Include="System.Device.Gpio" Version="1.2.0" />

Fails on both .Net Core 3.1 and .Net 5.0

After further investigation I have discovered that the hardware revision for the device is 9020e0, which is not recognised in the method System.Device.Gpio.RaspberryBoardInfo.GetBoardModel(); The hardware revision falls through the switch method and the generic driver is selected, which then throws the exception when opening the pin.

The enumerator System.Device.Gpio.RaspberryBoardInfo.Model does not have a value for the 3A+ board. The Raspberry Pi Model 3A+ is almost identical to the model 3B+. It has the same processor but only has 512MB of RAM, no ethernet card and a single USB port. Otherwise I can see no reason why the board should not be supported.

As a work around I have implemented the following when constructing the GpioController and my project is now working correctly on both the Raspberry PI Model 4B and Model 3A+

```csharp

                //we force the code to use the raspberry pi driver. This is because it does not recognize the 3A+ model I'm using
                var assembly = typeof(GpioDriver).Assembly;
                var driverType = assembly.GetType("System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver");
                var ctor = driverType.GetConstructor(new Type[]{});
                var driver = ctor.Invoke(null) as GpioDriver;

                ctl = new GpioController(PinNumberingScheme.Board, driver);

```

pgrawehr, 2020-11-29:
This problem has already been addressed, see #1270 . The current master should have this fixed.

Jumble0470, 2020-11-29:
I've seen the comment above, and I'm not sure it is correct. I've tested this with the latest release 1.2.0 and it is definitely not working. The revision number of 0x20e0 is not being recognized as a valid board. And the issue mentioned appears to reference a different model of pi (model 4B)

Ellerbach, 2020-12-23:
@pgrawehr and @krwq that should be now fixed in 1.3, right?

krwq, 2020-12-23:
I think so, @Jumble0470 can you try out 1.3 and confirm everything's working for you?

Jumble0470, 2021-01-29:
Sorry for the delay, I've not had time to look at this.

Jumble0470, 2021-01-29:
Sorry for the delay, I've had the chance to test this and it is working for me.

closed 2021-01-29 by Jumble0470