dotnet/iot #1330: Open and build projects in public version of Visual Studio (not the Preview version)

SongweiFu-ITEM, 2020-11-30:
Now the projects can only be loaded and built in Visual Studio Preview version. 
It would be nice to be able to load projects from Visual Studio 2019 (community / professional / Enterprise).

krwq, 2020-11-30:
cc: @joperezr

joperezr, 2020-11-30:
This should theoretically work as long as you have the .net 5.0 installed globally in your machine. I'll take a look to see if that is not the case.

SongweiFu-ITEM, 2020-11-30:
update from 16.5.4 to 16.8.2 solves the problem. no problem of loading the project from VS 2019.

pgrawehr, 2020-12-01:
Yes, updating to at least 16.8.0 is required to build with the final .NET 5.0 version.

Ellerbach, 2020-12-01:
So I guess this can be closed! Feel free to reopen if any other issue.

closed 2020-12-01 by Ellerbach