dotnet/iot #1344: BdfFont class cannot ever successfully call GetCharData()

negativeeddy, 2020-12-09:
**Describe the bug**

When calling [BdfFont.GetCharData()](https://github.com/dotnet/iot/blob/6537bbabf7424b0e0d561bd4a3bf98de35505b8f/src/devices/Common/Iot/Device/Graphics/BdfFont.cs#L135), the code will always throw an exception

**Steps to reproduce**
1. Load a BdfFont
2. Call GetCharData()

or

1. Run the [RGBLedMatrix sample](https://github.com/dotnet/iot/tree/master/src/devices/RGBLedMatrix/samples)
2. Press 1 to switch to scenario 1

**Expected behavior**

It should return the data for the character and the sample should run without throwing an exception. It looks like the throw should be in an else clause for that if statement. I made that change and the sample seems to run fine.

**Actual behavior**

It always throws an exception.

**Versions used**
.NET SDK (reflecting any global.json):
 Version:   5.0.100
 Commit:    5044b93829

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet/sdk/5.0.100/

Host (useful for support):
  Version: 5.0.0
  Commit:  cf258a14b7

.NET SDKs installed:
  5.0.100 [/home/pi/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 5.0.0 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 5.0.0 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

krwq, 2020-12-09:
@negativeeddy is the `fonts` folder in your current directory? (relative to the app you're running)

Note that fonts are not part of the library nor are resources. If you're using `dotnet publish` they won't automatically get copied to the output folder

negativeeddy, 2020-12-09:
yes the fonts folder is there. I'm just running the sample under the RGBLedMatrix folder as is. The scenarios with text rendering dont work because the exception is thrown.
 
I created PR #1355 so that you can see the issue more clearly. Today, that method will *always* throw an exception whether it finds the glyph data or not. After this change the sample app ran correctly for me.

krwq, 2020-12-09:
Yes, seems we've regressed this recently:
https://github.com/dotnet/iot/pull/1243/files#diff-50e44f04eb83b207d2af985fc78c6c93c231d769dff7ade393fa5d3bea76ce98R138
cc: @richlander (if this was automated/semi automated then we might wanna take a second look there)

PR #1355 (2020-12-10): only throw exception if glyph data not found
Fixes #1344
- Problem: All calls to BdfFontGet.GetCharData() will throw an exception, regardless of whether the font was loaded properly or not.
- Fix: only throw an exception if the glyph data cannot be located
closed 2020-12-10 by krwq