dotnet/iot #1357: Should WeatherHelper use extension methods?

krwq, 2020-12-09:
After merging https://github.com/dotnet/iot/pull/1356 we should be able to implement WeatherHelper entirely as extension methods to Temperature/Pressure/RelativeHumidity,

Do you think this would be useful? If so we should decide which ones will go where and if i.e.:
CalculateDewPoint should be on both temperature and RelativeHumidity or just humidity. If we decide with just RelativeHumidity then we should consider changing argument order as currently humidity is second

cc: @pgrawehr @Ellerbach @joperezr @MarkCiliaVincenti
Ref https://github.com/dotnet/iot/issues/1339

pgrawehr, 2020-12-09:
I'm not sure this is a good idea. For the WeatherHelper functions, you always need (at least) two imput values, and they're equaly important, so it would be hard to say which one to prefer. And having both may get confusing.

MarkCiliaVincenti, 2020-12-09:
Agreed with @pgrawehr. In fact I have a project where I am using these NuGet packages just to use WeatherHelper. Given that they're not big enough to warrant their own standalone package, leaving them isolated is probably the best way to go here, not just for the reasons pgrawehr brought.

krwq, 2020-12-10:
Sounds good. The reason I opened the issue is because I wasn't sure myself. Closing in this case.

closed 2020-12-10 by krwq