dotnet/iot #1363: This driver is generic so it can not perform conversions between pin numbering schemes.

netleike, 2020-12-11:
我更新了最新System.Device.Gpio 库后提示
This driver is generic so it can not perform conversions between pin numbering schemes.

GpioController controller = new GpioController(System.Device.Gpio.PinNumberingScheme.Board);
                if (!controller.IsPinOpen(pin))
                {
                    controller.OpenPin(pin);
                    controller.SetPinMode(pin, System.Device.Gpio.PinMode.Output);
                }
              
                controller.Write(pin, System.Device.Gpio.PinValue.High);
                Thread.Sleep(1000);
                controller.Write(pin, System.Device.Gpio.PinValue.Low);
                Thread.Sleep(1000);
                return Content("ok");

pgrawehr, 2020-12-11:
What hardware are you running on? 

Except for the Raspberry Pi, you cannot use PinNumberingScheme.Board. Use PinNumberingScheme.Logical instead.

netleike, 2020-12-12:
Thank you the problem has been solved

closed 2020-12-12 by netleike