dotnet/iot #1365: System.device.Gpio version 1.2.0 DTH11 ERRO

netleike, 2020-12-12:
About System.device.Gpio version 1.0.0
GpioController controller = new System.Device.Gpio.GpioController
High and low level diagram, send signal as shown in the figure
DTH11 reading results are normal.
But I'm using system.device.Gpio version 1.2.0
Abnormal DTH11 reading results, oscilloscope output image normal
Since I am not an electronic major, I cannot check the cause of this problem. Please help to study it.
thank you

关于System.Device.Gpio 版本1.0.0
GpioController controller = new System.Device.Gpio.GpioController
高低电平图，发送信号如图所示
读取DTH11结果正常。
但是我用System.Device.Gpio 版本1.2.0
读取DTH11结果异常，示波器输出图像正常
由于我不是电子专业，无法排查该问题出现原因，请大家帮忙研究下。
谢谢

pgrawehr, 2020-12-13:
Can you be a bit more specific about what you mean by "abnormal reading results"? Are the values just wrong? or no readings at all?

microhobby, 2020-12-19:
@netleike it would also be good to share what hardware and operating system you are using, so we can try to test and try to reproduce.

closed 2021-01-03 by netleike