dotnet/iot #1373: [Thoughts] Future of Repo

shaggygi, 2020-12-18:
I recently came across [an announcement](https://www.thurrott.com/smart-home/245199/google-kills-android-things) Google was deprecating one of [their projects](https://developer.android.com/things) that can be compared to System.Device.  This brings up the question if System.Device will have a similar fate one day.

I really like this project and what it offers for the .NET ecosystem.  IMHO, it is just scratching the surface on possibilities and could have a great future (especially as .NET 5+ becomes more powerful/performant and more devices/architectures are supported).  This is why I was one of the early adopters and contributors.

I still watch this space very closely, but have not been involved as much over the past year for a few reasons.  As usual, personal life and work (that pays the bills) has priority and there is only so much time afterwards to allocate to help (but hey, that's the open-source life isn't it 😄).

I'll also be completely honest that I haven't been involved because this project doesn't appear to have the support it needs in order to persuade developers to "join the party" and make this space grow.  There are fundamentals things that still need to be addressed such as some standards on device bindings (naming conventions, etc.), fundamental decisions to be updated (like separating board drivers from core API, etc.) and core features/issues that need added/fixed (low level protocol bugs, etc.).  In addition, there doesn't seem to be many resources dedicated from Microsoft.  This space is not even mentioned in the .NET conferences now (or at least doesn't have its own sessions to show off features/roadmaps).

While I can't provide specifics, I've been working on a project for my company that could utilize this API.  However, I'm very hesitant as I don't want to recommend we go down the path of using in our process to find out a couple years later it is now deprecated (like Google announced this week). 

Please don't take that I'm poo-pooing on this project as I've been a cheerleader from day one (still am, just at a distance right now).  Just throwing this out to see where Microsoft and some of the heavier contributors see for the future of this awesome offering from .NET.

Thanks for reading.

krwq, 2020-12-18:
cc: @richlander

microhobby, 2020-12-18:
@shaggygi your concern is valid, but I think the Google project was much more ambitious than what we have here. They worked on a secure BSP, delivering bootloader, Kernel, secure chain and etc for hardware and also an OTA service. Only the part of having a custom BSP for the hardware to run their Android "Things" was already a shot in the foot. This would only be compatible with the hardware they work with or partners.

What we have here in the `dotnet/iot` is generic enough to run on Windows and several and several platforms based on Linux, I say several because every day I discover some new platform with a CPU vendor that I have never seen in my life being released.  And if it runs Linux it must have already implemented the Kernel interfaces that we use here and it will "automatically" be compatible with `dotnet/iot` API.

I have recommended the library whenever possible and I try to help with which I have the knowledge, but it is not always that we can be active for the problems you have already listed. It's great to see that Microsoft has maintained the project here within the `dotnet` context and this gives us a lot of confidence to use `dotnet/iot` in production, not only in "maker" projects. I really hope that they continue with the great work that has such great use.

With the growing demand for devices on the edge, IoT, IIoT, and automation projects, I don't see why the project would be discontinued, but if that happens, it is also open source. So I hope the community will continue with the work. But I really hope it continues as it is ...

And here is my acknowledgment and thanks to the team, maintainers, and contributors for the great work here 🥰

pgrawehr, 2020-12-19:
I don't know about the google project, but this here is at least open source, so even _if_ Microsoft decided to drop support, you could still continue to use it for your projects and even continue developping it on your own.

shaggygi, 2020-12-19:
Also just happen to come across this today.  [Growing the .NET ecosystem](https://github.com/microsoft/dotnet/blob/master/docs/ecosystem-issues.md#feedback-critical-opinions-on-the-microsoft-oss-ecosystem)

> We also need to change our approach when we create net-new technologies for which there is no ecosystem yet, as is the case for **GPIO-based IOT libraries for .NET Core**. Instead of us building everything, we should create new projects in such a way that we're not the sole maintainer and actively seek out and invite external contributors to take part in long term ownership and maintenance. This also includes our naming patterns. For example, it has become the norm that we name packages and namespaces with Microsoft, which isn't very friendly for creating multi-vendor projects.

This is why I believe it is important the fundamentals are figured out first so the community can help it grow.

richlander, 2021-01-06:
Sorry the delay. I took a lot of time off for the EOY holidays.

Great topic. We had significant investment in this repo for .NET Core 3.0 and then very little for 5.0. We had to starve this project to get other things done. We intend to invest more in 6.0 and than we did for 5.0, although not dramatically so. Our plan going forward is that this project remaining relevant, in a few ways.

* Issues and PRs are regularly triaged, such that the community can be effective and active and is encouraged to do so by witnessing action on the repo. This has already started, if you look at the repo.
* Provide solutions to architectural / lower-level problems/needs that are only really appropriate if the repo maintainer is actively engaged on this problems.
* Provide more visibility for the project, on the dotnet blog and other places. This includes improving the demos and samples on the repo.

We welcome involvement on these topics. Talk to us if you want to get more involved on something specific.

Stepping back, a lot of what I wrote is about GPIO (and friends). There is also Linux and Arm as related but separate domains. We are very focused on both of those, particularly [Arm64](https://devblogs.microsoft.com/dotnet/arm64-performance-in-net-5/). We intend that all of our Linux and Arm64 investments accrue to the IoT scenario.

We are continuing to maintain Linux Arm32 but are not investing to make it better. 95% of our Arm investment is going into Arm64. This includes Windows Arm64 and the Apple M1 chip. We have dropped support for Windows Arm32. I see no plan to drop support for Linux Arm32. When the Raspberry Pi foundation moves to make Arm64 their default offering, that may change. That's not a pre-announcement, but an indication of what our long-term Linux Arm32 support might look like and what it might be hinged on.

We also have no current places for Armv6. That could change, but it is not a priority for us. We certainly need to deliver everything I spoke to above first.

Ellerbach, 2021-02-18:
[Triage] @shaggygi, can you confirm that the answer is sufficient so we can close this issue?

closed 2021-02-18 by shaggygi