dotnet/iot #1379: Documentation/raspi-i2c.md should show I2cBus example

krwq, 2020-12-22:
https://github.com/dotnet/iot/blob/master/Documentation/raspi-i2c.md should show I2cBus example (scanner in the bottom might be a good use case)

shaggygi, 2020-12-24:
Might want to add the Device Tester I2C section while updating.

There is [a section](https://github.com/dotnet/iot/blob/master/Documentation/raspi-i2c.md#bonus-write-your-own-i2c-scanner) on creating a device detection.  We have a [I2cDetect command](https://github.com/dotnet/iot/blob/master/tools/DevicesApiTester/Commands/I2c/I2cDetect.cs).

pgrawehr, 2020-12-25:
The Device Tester tool should be updated after my board PR is merged, there are a few new possibilities there.

Ellerbach, 2020-12-27:
Let' give the few possibilities, it's always a good way to give examples.