dotnet/iot #1381: We should enable SA1201, SA1202, SA1203, SA1204, SA1214 and SA1215

krwq, 2020-12-22:
See https://stackoverflow.com/questions/150479/order-of-items-in-classes-fields-properties-constructors-methods

This would improve readability in the code in general

Ellerbach, 2020-12-23:
I'm not super in favor of those rules. For the bindings I've developed and in general when I'm coding, I regroup elements logically and not by their access type. If you look at the long bindings, they all have regions, grouping things logically. It makes it millions times easier to maintain than if all those functions, properties would be grouped by access level.
I'm find with the constant on the top, the global variables on the top, the constructor on the top but for the rest, I would let the logical flow of grouping by features.

krwq, 2020-12-23:
fair enough, we should only pick those we all feel comfortable with

pgrawehr, 2020-12-25:
I think rules SA1201, 1203 and eventually 1214 and 1215 are fine, but SA1202 and SA1204 really don't help in adding readability. They force that all static methods must come before all non-static methods, and private methods must come after public methods etc. Besides that one generally places a private helper method next to where it's used, this has also the downside of requiring that a method must be moved within the file when it's visibility changes.

Ellerbach, 2021-02-18:
[Triage] We will investigate the following:
* SA1201 - ? depends on number of changes
* SA1203 - ok
* SA1204 - ? depends on number of changes
* SA1214 - ok
* SA1215 - ? depends on number of changes

krwq, 2021-02-19:
* [x] SA1201 - Not convinced about it, seems more noisy than worth it, I didn't find the order super obvious
* [x] SA1203 - ok - already enabled
* [x] SA1204 - I found the code easier to read in most cases
* [x] SA1214 - ok - already enabled
* [x] SA1215 - ? depends on number of changes - no errors so assuming this was intuitively done correctly

closed 2021-02-26 by msftbot[bot]