dotnet/iot #1391: Stylecop SA1008 creating issues with VS2019 autoformatting

Ellerbach, 2020-12-23:
**Describe the bug**

See the simple code below. Do `ctrl + k + d` for autoformatting in VS2019. and Stylecop is not happy because of rule SA1008

[image]

**Steps to reproduce**

```csharp
var platformChoice = Console.ReadKey();
Console.WriteLine();

if (platformChoice is not object or { KeyChar: not ('1' or '2') })
{
    Console.WriteLine("You have to choose a valid platform");
    return;
}
```

**Expected behavior**

Autoformatting should respect the rules or rules needs to be adjusted.

**Actual behavior**

Create an error which needs to be manually adjusted every time.

**Versions used**

Microsoft Visual Studio Enterprise 2019
Version 16.8.3

pgrawehr, 2020-12-25:
Not sure, but it is possible that the settings in `.editorconfig` contradict the StyleCop rules in some points.

Ellerbach, 2021-02-18:
[Triage] There are still few edge cases where there are conflits. We'll let this one open to make sure we don't forget to fix all those edge cases.

Ellerbach, 2021-03-25:
[Triage]This seems to be fixed for now.

closed 2021-03-25 by Ellerbach