dotnet/iot #1399: Where is the API docs?

GF-Huang, 2021-01-01:


Ellerbach, 2021-01-01:
@GF-Huang can you be a bit more precise please? What are you looking for exactly?
We do have documentation for every binding with a sample. Check any of the /src/devices
For System.Device.Gpio we do have as well in the documentation for example for SPI: https://github.com/dotnet/iot/blob/master/Documentation/raspi-spi.md

krwq, 2021-01-03:
Do you mean https://docs.microsoft.com/en-us/dotnet/api/?view=iot-dotnet-1.1 ?

GF-Huang, 2021-01-03:
yes，thanks

pgrawehr, 2021-01-03:
@krwq: He cool, didn't know that this was "officially" published. We should make sure we link there from the main Readme and also make sure the documentation is updated regularly.

krwq, 2021-01-04:
Yes, https://github.com/dotnet/iot-api-docs is the repo. I'll need to ask people who created it to update it since it's currently only documenting version 1.1 (hopefully set auto-update on the repo).

Ellerbach, 2021-01-04:
And we could as well add properly all the README.md and all the rest of the documentation. I've been using quit a lot DocFX and also wrote a TOC generator so I may have a look at this more deeply!

Ellerbach, 2021-02-18:
[Triage] The question is answered. The process to update it is on its way. Feel free to reopen it if you have specific elements or you don't see anything updated in the next few weeks.

closed 2021-02-18 by Ellerbach