dotnet/iot #1400: Add Zigbee Lora communications

netleike, 2021-01-03:
IOT domain is dominated by Zigbee Lora. Currently, there is no relevant library at.NET Core. Could you please add lora ZigBee related library?
Lora SX1278 and other chips!
thank you

archanox, 2021-01-07:
I've got a couple of SX1262s in the mail, which I hope to get running when they arrive.

Are there any pointers on how to set them up and submit a PR?

Ellerbach, 2021-01-09:
@netleike Please check https://github.com/Azure/iotedge-lorawan-starterkit, that should help, if you're looking at connecting Lora devices to Azure Iot Hub.
That does allow you to have multiple gateway, a central device management, OTA on boarding, create your own message decoder and more!

Ellerbach, 2021-02-18:
[Triage] Support seems beyond scope for .NET IoT. Most of the time, those chips are connected to a very small micro controller as the key scenarios for LoRa is long range, low cost communications. ZigBee can be slightly different as it can be connected all the time. That said, same, they're mainly used with low end, low consumption micro processors.
If anyone wants to add the support in .NET IoT, that's of course possible.
So closing this issue, feel free to reopen if you have some more specific elements.

closed 2021-02-18 by Ellerbach