dotnet/iot #1401: [Question] How to write data with REGISTER using I2C

scegg, 2021-01-04:
Hi.

I need to translate a code block written in python to C#.

The python block is
```
bus = SMBus(1)
address = 0x3C
bus.write_byte_data(address, 0x40, value)
```

Here is my C# block
```
_i2cBus = I2cBus.Create(1);
_i2cDevice = _i2cBus.CreateDevice(0x3C);
_i2cDevice.Write(value); //Problem
```

Problem is how to transfer the REGISTER (0x40)?

Thanks.

scegg, 2021-01-04:
Related code:
smbus2 for python: https://github.com/kplindegaard/smbus2/blob/master/smbus2/smbus2.py
your code I guess (I'm using 1.3 version from Nuget): https://github.com/dotnet/iot/blob/release/3.0/src/System.Device.Gpio/System/Device/I2c/Devices/UnixI2cDevice.Linux.cs

krwq, 2021-01-04:
The easiest way will be to look at existing devices, i.e.:

https://github.com/dotnet/iot/blob/master/src/devices/Hts221/Hts221.cs#L51-L79

So what most people do is first define an enum, i.e.:
https://github.com/dotnet/iot/blob/master/src/devices/Hts221/Register.cs

Then they cast it to byte when transmitting over the wire.
You can also send bytes directly but code is cleaner when using enums

scegg, 2021-01-04:
Thanks @krwq .

So all I need is:
1 Create an array of 2 bytes, containing Register and the Value.
2 Send the array using Write instead of WriteByte.

Am I correct?

Reading the source code of the UnixI1cDevice, it may includes other data to be encoded like length of data.

krwq, 2021-01-04:
@scegg usually yes. It all depends on the device in the end, some may have specific requirements, you will need to refer to a spec. Some devices might be ok with using WriteByte twice instead but most of the time it will need to be a single Write with two bytes.

scegg, 2021-01-04:
@krwq Thanks a lot. Let me close this issue for now.

closed 2021-01-04 by scegg

krwq, 2021-01-04:
Sure, let me know should you have any more questions!